OctoPrint/OctoPrint #4107: Webcam stream not loading

charlespick, 2021-04-22:
### The problem

The webcam stream isn't loading even though it's configured properly. Clicking the configured url link and the test button in settings work. 

GitHub keeps deleting my images so here's an imgur link I guess https://imgur.com/a/7wgjapi 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.5.3

### Operating system running OctoPrint

Raspian 10 Buster NOT octopi

### Browser and version of browser, operating system running browser

Chromium Edge, Brave, Firefox, mobile Safari

### Checklist of files to include below

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

octoprint.log: https://drive.google.com/file/d/1ibIms_IVGcAKo-Kkd9Gip-VR1jH795kq/view?usp=sharing
chrome console log: https://drive.google.com/file/d/18AWleeEC2xYrhQuFnauxox7PFFtmkWTk/view?usp=sharing

Xadiant, 2021-04-24:
I am having this exact same issue.

Running Octoprint 1.5.3 installed using the manual steps on existing Pi4 running Raspberry Pi OS.

cp2004, 2021-04-24:
Please also upload the logs, JS and make sure it still works just going to the URL on its own.

Xadiant, 2021-04-24:
@cp2004 Yes, if I click the URL that the webcam stream box says does not work, my .mjpg stream works fine immediately.
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/6370378/octoprint.log)

Xadiant, 2021-04-24:
I just figured it out, looking at the log from my camera host (python script) I saw the line:
`192.168.18.133 - - [24/Apr/2021 15:15:57] code 404, message Not Found
192.168.18.133 - - [24/Apr/2021 15:15:57] "GET /stream.mjpg?1619291757638 HTTP/1.1" 404 -`

So I modified my camera server to ignore anything after the .mjpg (the number that octoprint adds)

Why does octoprint append this "?1619291757638" to the stream path?

jneilliii, 2021-04-25:
> Why does octoprint append this "?1619291757638" to the stream path?

This is what's known as a cache buster... to force a reload I'd the file bypassing any local browser cache, very common.

cp2004, 2021-04-25:
@charlespick do you have the same issue here, with the additional cache-buster? What webcam server are you using?

You could also post your reverse proxy configuration (is it haproxy?).

Xadiant, 2021-04-25:
> > Why does octoprint append this "?1619291757638" to the stream path?
> 
> This is what's known as a cache buster... to force a reload I'd the file bypassing any local browser cache, very common.

Cool, I feel very lucky to have stumbled upon the actual issue here.  If there was a checkbox in the webcam/stream settings of octoprint to enable/disable cache busting I probably would have toggled it and figured it out within a few minuets of intial setup.  Then again, if we are the first two people to encounter this issue then it may not be worth the change.

cp2004, 2021-04-25:
You aren't the first, there's a bunch of requests here already, but so far no one has done anything about it. It shouldn't be too difficult to implement, I have it on my list of potential PRs.

If anyone wants to give it a go, the relevant setting needs plumbing in and adjusting here:

https://github.com/OctoPrint/OctoPrint/blob/6c0176455bd80551bcf473e424cd56f5aab04254/src/octoprint/static/js/app/viewmodels/control.js#L696

Maybe different for the timelapse as well, if it covers that. Just make sure you don't have a cache issue otherwise.

anonymousThey, 2021-05-04:
I've run into this one as well. Not been a huge blocker, but it would be nice to have a cache_busting toggle somewhere. 
I added a $50 bounty if anyone is interested!

foosel, 2021-05-04:
PR #4124 from @cp2004 merged, will ship with 1.7.0.

smizles, 2021-07-01:
Question, I'm having a similar issue with the exception the URL doesn't work. However, 1 out of 7 times the webcam randomly loads. It was working flawlessly for the 1st 24 hours. I've triple-checked the connections.

I'm running a Pi Camera v2 on a Raspberry Pi 4 2GB. I setup Octopi according to the official instructions. Additional info: OctoPrint 1.6.1, Python 3.7.3, OctoPi 0.18.0

FYI, I'm able to SSH in, take a still with the "raspistill" command, download it with WinSCP, and the image is a "live" image as in it was a recent subject that appeared in front of it not some sort of artifact. I've tried rebooting numerous times including "safe mode", and I just recently scrapped everything and reflashed my SD card with Octopi only to have the same issue once I ran the initial wizards again.

Please let me know if my problem is related to this or if I should open a separate issue.

foosel, 2021-07-01:
That sounds unrelated and rather like an issue with the webcam server than OctoPrint embedding the stream, and therefore outside of the scope of this repository. Best get in touch on the community forums.

PR #4124 (2021-05-04): ✨  Add option to disable cache busting on webcam URL
#### What does this PR do and why is it necessary?

Adds an advanced option to disable the cache-busting `?1620136984425` or similar added to the webcam URL.

This causes issues with several webcam streamers, notably [ustreamer](https://github.com/pikvm/ustreamer) and also with some reverse proxy setups.

#### How was it tested? How can it be tested by the reviewer?

Manually on my instance, changing the settings about.

#### Any background context you want to provide?

See tickets

#### What are the relevant tickets if any?

Resolves #4107, [... cut]