OctoPrint/OctoPrint #4108: OctoPrint 1.5.3 causes OctoPrint to stop working

victaru, 2021-04-24:
### The problem

After installing the latest 1.5.3 update, the server is unable to start. Re imaged the SD Card with those from https://octoprint.org/download/ and managed to get OctoPrint back

Printer: Ender 3 Pro
Pi: Raspberry Pi 3B+

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Unable to restart

### Version of OctoPrint

1.5.3

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Ender 3 Pro

### Browser and version of browser, operating system running browser

Chrome

### Checklist of files to include below

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Re-imaged the drive, no log.

cp2004, 2021-04-24:
There is really nothing we can do about this with no information. There is no widespread issue with 1.5.3, with over 87,000 active installs ([source](https://data.octoprint.org)) in the last 30 days. Something likely went wrong in the update process with your install.

Should you run into an issue with your OctoPrint install, you can seek support from the community either on https://community.octoprint.org or https://discord.octoprint.org. It's likely we would have been able to help you figure out the issue and get it resolved, properly diagnosing the issue and working for a resolution. But as it stands, there is no information to do anything here.

closed 2021-04-24 by cp2004

victaru, 2021-04-25:
Hi, I would love to provide logs, but for some reason I can't do the bar "¦" symbol in the console to do the troubleshooting steps. For some reason it won't come out from my various keyboards.

victaru, 2021-04-27:
right so i managed to putty in and i found the Octoprint folder empty with only the README file. it seems that during the update process it uninstalled the old version but failed to install the new one.  so i went through the manual install guide and got the latest version running.

cp2004, 2021-04-27:
If you still need help please seek support on the Community forums or the Discord server.

The folder at `~/OctoPrint` has always been empty, it is a space for a local checkout. OctoPrint is directly installed using pip and has been for years.