OctoPrint/OctoPrint #4109: Minor Bug: Update All button active when nothing needs updating

KenLucke, 2021-04-25:
### The problem

If Software Update shows everything up to date, the "Update All" button is still active, clickable, and brings up the normal dialogs, even though there's nothing to update.   Don't think this is a new issue, just reminded of it when it happened today.

The "Update All" button should be dimmed and unclickable if there is nothing to update.

[image]
[image]
[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.0rc3

### Operating system running OctoPrint

Mac OS 10.15.7

### Printer model & used firmware incl. version

n/a

### Browser and version of browser, operating system running browser

n/a

### Checklist of files to include below

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Tried to upload system info bundle - this box will not accept them.

https://www.dropbox.com/s/6t5t3yr0lm0t1os/octoprint-systeminfo-20210424192059.zip

cp2004, 2021-04-25:
> Tried to upload system info bundle - this box will not accept them.

This is a bug with the new GitHub issues forms. As they say, they are alpha, so expect bugs I guess. I can't seem to get any files I put there to stick.

I think only @foosel is the one with a contact to report bugs with them.

foosel, 2021-04-26:
> Tried to upload system info bundle - this box will not accept them.

I've reported this upstream. Looks like uploads are possible (even though the UX could use some improval there as it only speaks about images) but once you focus on anything that isn't the textarea you uploaded something to the upload gets erased from the markdown.

foosel, 2021-04-26:
And on the actual bug this ticket is about: I just fixed that on `staging/maintenance`, but it will be until a possible bug fix release for 1.6 becomes necessary or 1.7 for this to be shipped. Currently 1.6.0rc3 is looking good enough to release (looking at going for that possibly tomorrow) and this is too minor to justify another RC round.

KenLucke, 2021-04-26:
> And on the actual bug this ticket is about: I just fixed that on `staging/maintenance`, but it will be until a possible bug fix release for 1.6 becomes necessary or 1.7 for this to be shipped. [...]

I kinda figured that would be the way to go, no sense stopping the release for such a niggling bug.  I just wanted to be in the queue.

closed 2021-04-27 by foosel

foosel, 2021-04-27:
Well... that accidentally made it into 1.6.0 after all O_O I have no idea how that happened, must have committed on the wrong branch. That's... suboptimal, but well...

commit 0cf06a60 (2021-04-26): 🐛 SWU: Disable update all if no updates

Closes #4109