OctoPrint/OctoPrint #4115: [Request] drag or pan objects in gcode viewer

tx0h, 2021-04-28:
### Is your feature request related to a problem? Please describe.

hello,

i just updated to 1.6.0 right now and in the previous release in the gcode viewer i was able to zoom into the object and pan or drag with the LMB and move the mouse to parts of the object. it seems that this feature is gone in 1.6.0. can we get this feature back?

thanks,
tx0h

### Describe the solution you'd like

just roll back this change.

### Describe alternatives you've considered

i could just install the old version. this feature was important to me.

cp2004, 2021-04-28:
It works fine for me -  I can drag the gcode viewer about to look at any part of the bed.

Please try in safe mode, screenshot the settings you have checked below the gcode viewer and upload your gcode file so we are trying the same thing.

tx0h, 2021-04-28:
hi,

i can't restart in safe mode right now, because a print is running. i also didn't reboot after the upgrade.

i will attach the screenshot now and the gcode file.

thank you for your info!

tx0h, 2021-04-28:
no, not working. not in seamonkey, not in chromium..

jneilliii, 2021-04-28:
This issue seems to be tied to center origin. What if you change that in your printer's profile to lower left instead of center?

foosel, 2021-04-28:
Please read the change log, there's NOTHING about removing this functionality in there. So this is actually a bug report in disguise and should have come with a filled out template, safe mode testing and systeminfo bundle included.

Can confirm however that there is an issue if the printer's profile is set to center origin.

tx0h, 2021-04-29:
> This issue seems to be tied to center origin. What if you change that in your printer's profile to lower left instead of center?

wow! how did you know?! at least a work around. ty!

linuxdan, 2021-05-02:
While you are tinkering in the viewport, could you move the vertical scroll bar over so that it does not block the model ?
Or does that need a separate issue opened up ?

foosel, 2021-05-03:
Please open a new bug report for that. I don't see anything blocking anything and besides, different thing, different issue.

foosel, 2021-05-03:
Fix ready for 1.6.1

closed 2021-05-10 by foosel

foosel, 2021-05-10:
1.6.1 got released.

commit 19c39588 (2021-05-03): 🐛 GCV: Fix dragging w/ center origin & bed changes

Minor refactoring to get rid of three problems. Not only is #4115 as
introduced by 886c3d9449c3ddd4e5a9dd73a1397a564996db4c fixed but changes
to the bed dimensions after printer profile change now trigger a proper
viewport reset (zoom & translation) and there's now also a button to
reset the viewport to a sensible default should you get completely lost
in the viewer canvas.

Closes #4115