OctoPrint/OctoPrint #4117: GCode Viewer Visualisation Problem

Speedy75, 2021-05-01:
### The problem

The visualisation in GCode viewer ist not correct. The print is OK.
See gcode file (zip) on Layer 43 to 47 and 49
And screenshot

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.0

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Ender 3 V2

### Browser and version of browser, operating system running browser

Chrome

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[image]

[Black_PETG.zip](https://github.com/OctoPrint/OctoPrint/files/6410207/Black_PETG.zip)
[octoprint-systeminfo-20210501215703.zip](https://github.com/OctoPrint/OctoPrint/files/6410208/octoprint-systeminfo-20210501215703.zip)

cp2004, 2021-05-01:
Confirmed, haven't poked into the actual gcode file to see what it contains but alternative gcode viewers don't show this same issue. Was present in 1.5.x as well, not a regression.

foosel, 2021-05-05:
Confirmed *only* on Chrome:

[image]

Everything is fine on Firefox:

[image]

Which makes this *very* weird indeed. Maybe a different rounding behaviour somewhere?

foosel, 2021-05-05:
@Speedy75 Sidenote, though unrelated to this ticket:

```
env.plugins.pi_support.throttle_state: 0x50000
```

You have an intermittent undervoltage issue on your pi.

Speedy75, 2021-05-05:
> @Speedy75 Sidenote, though unrelated to this ticket:
> 
> ```
> env.plugins.pi_support.throttle_state: 0x50000
> ```
> 
> You have an intermittent undervoltage issue on your pi.

Oh, thanks. i will check cooling fan and power supply

foosel, 2021-05-05:
```
M82
G92 E0
G28
G1 Z1
G1 X173.595 Y103.9 E247.16716
G3 X173.600 Y126.097 I-105613.507 J39.645 E248.20080
G1 X169.552 Y126.098 E248.38933
```

Minimum example with a reproduction. The `G3` is the problematic issue. Haven't yet figure out why though.

@Speedy75 could you also share the non arc weldered version of the file?

Speedy75, 2021-05-05:
not for this modell, i dont have it. i try to resclice, check visualisation and save both files if i can find error

foosel, 2021-05-05:
So that is *weird*. I've created a new debug mode that visualizes arcs as they are drawn basically as the pizza slices they are, with red dashed lines. E.g.:

[image]

The problematic GCODE renders in Firefox like this:

[image]

and in Chrome like this:

[image]

Note how in Firefox the arc stops where it's supposed to stop, whereas in Chrome it overshoots. The arc parameters are absolutely identical in both cases (I logged and compared them). 

@FormerLurker have you possibly seen anything like this ever while debugging arc welder stuff?

Speedy75, 2021-05-05:
i resliced the same modell, and have no errors in following layers: 62, 63, 64, 66

both files here
[ARC_WELDER.schaukel_v4.zip](https://github.com/OctoPrint/OctoPrint/files/6426751/ARC_WELDER.schaukel_v4.zip)
[NOARC_WELDER.schaukel_v4.zip](https://github.com/OctoPrint/OctoPrint/files/6426752/NOARC_WELDER.schaukel_v4.zip)

foosel, 2021-05-05:
Apparently there are some rounding errors present in Chrome's implementation of the canvas context's `arc` method. Replacing it with an approximation using bezier curves makes the issue disappear:

[image]

However, that also makes the rendering bog slow in comparison. I'm a bit unsure right now whether I should go for the workaround (limited to Chrome only) or just chalk this up to a browser bug that happens under very specific circumstances (HUGE arc radii with teeny tiny angles) and let it be as is. Or, there's option C: Add the workaround as a debug option that can be enabled by the user if need be.

jneilliii, 2021-05-05:
I'd opt for that last option, where the user can enable/disable it since I'm a Chrome user and don't care so much about the accuracy of the gcode viewer rendering and would prefer the better performance.

foosel, 2021-05-06:
Good news, it isn't actually slower after all. I must have had some debug logging still in place that messed up the check. I've still made it an option that on chrome defaults to on, but can be disabled on a per browser basis (and which will also be invisible on other browsers):

[image]

Works surprisingly well:

[image]

foosel, 2021-05-06:
And ready for 1.7.0.

commit e1f8a82d (2021-05-06): 🥅 GCV: Work around for Chrome bug

Chrome's canvas arc method seems to produce wrong results in case of
extremely large radii combined with extremely small enclosed angles. So
in case of Chrome we'll now default to a replacement function
approximating the arc with bezier curves. This can however be disabled
via the advanced options.

Additionally, another debug visualization option was added that shows
the arcs with their center points and segment borders.

Closes #4117
closed 2021-10-11 by foosel