OctoPrint/OctoPrint #4118: [Request] Option to *NOT* Simply3D Scripts when simply plug in is installed

tfaz099, 2021-05-02:
### Is your feature request related to a problem? Please describe.

The problem is that the Simply3D Script via a web interface is not always feasible if the net is down.  Need a way to override this to allow editing of the scripts directly in octoprint

### Describe the solution you'd like

"add override switch to edit script and not allow simply3d to update it"

### Describe alternatives you've considered

none

cp2004, 2021-05-02:
Not an issue with Core OctoPrint, a plugin issue. Please report issues with 3rd party plugins to the plugin author via their own contact channels.

closed 2021-05-02 by cp2004

tfaz099, 2021-05-02:
I disagree that it should be left on the third party plugin owner.  It is Octoprint allowing it be overridden and locked.  If they allow the override, we should be able to turn that option off within Octoprint.  it is a security issue also and security is on the onis of octoprint, not simply3d

cp2004, 2021-05-02:
OctoPrint is not _allowing_ the plugin to do anything here, this is just how JavaScript works. It does this itself, and there is nothing you can do to stop it from OctoPrint's point of view. If it is such a security issue that a plugin can change the UI, I guess we might have to remove the plugin interface. Now have a think about how many people that might upset.

This single line of JavaScript would remove the entire UI:
```js
$('body').remove()
```

And there is nothing you can do to stop it - if a plugin entered this, you now have no UI.

> it is a security issue also and security is on the onis of octoprint, not simply3d

The security of your instance is not compromised, you are not vulnerable to anyone attacking your OctoPrint install and causing issues.