OctoPrint/OctoPrint #4119: Refresh filemanager cause freeze web

devildant, 2021-05-02:
### The problem

hello, here is a bug that appeared with octoprint 1.6.0, if you click refresh file manager while sd printing the interface hangs until the printer or printing stops

1: launch print from sdcard (octoprint interface)
2: wait for the bed temperature to start to heat up
3: click on refresh button on sidebar filemanager
[image]
4: refresh browser 
5: the interface is block (all ajax with status pending)

to recover the interface you must turn off the printer

OctoPrint 1.6.0 
Python 2.7.16 
OctoPi 0.17.0

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.0

### Operating system running OctoPrint

octopi 0.17

### Printer model & used firmware incl. version

cr10s fw th3d marlin 1.9

### Browser and version of browser, operating system running browser

all

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[OctoPrint-1.6.0.zip](https://github.com/OctoPrint/OctoPrint/files/6412252/OctoPrint-1.6.0.zip)

cp2004, 2021-05-02:
Could you upload the systeminfo bundle?

devildant, 2021-05-02:
> Could you upload the systeminfo bundle?

Hello, already upload, inside OctoPrint-1.6.0.zip

devildant, 2021-05-02:
uploaded :)

devildant, 2021-05-02:
I think a control has disappeared in the filemanager api.
example:
if (!isSdPrint)
{
   checkSdFile()...
}
it's just a guess

foosel, 2021-05-03:
Your guess is close to the truth. 1.6.0 saw a change in how the refresh button works with regards to the SD files. Before it would just used the cached result, now it will request a fresh list, and block for something like a minute or so (unsure about the precise timeout right now) to query that from the printer. I could imagine your server didn't actually lock up for good, it was just sitting in that timeout. What I don't understand is that stuff is blocking to begin with, `M20` should still be sent even if a print is going on. The fix should still be simply to NOT refresh if a print is going on of any kind.

foosel, 2021-05-03:
Well... the timeout works alright, but it was not set to 10s but to 10000s because I misread the API documentation and thought I was supposed to specify milliseconds 🤦‍♀️

foosel, 2021-05-03:
Fix ready for 1.6.1

devildant, 2021-05-03:
> Well... the timeout works alright, but it was not set to 10s but to 10000s because I misread the API documentation and thought I was supposed to specify milliseconds 🤦‍♀️

haha, this happens :) this explains why after 20min I still did not recover the interface ;p

yes do not use the refresh button :)
except that I had one of my plugins which triggered the refresh via js on each load (to add a button on each file present on the pi, I unfortunately could not find a cleaner way to do it)

it took me a while to isolate the plugin and target the problem
I ended up opting for a refresh only if the status is operational

ps: once again sorry for not having participated in the release candidate, I was overwhelmed by the multiple updates that arrived with the containment (php, angular, electron, node, api ...)

and thank you for the responsiveness :)

foosel, 2021-05-10:
This is fixed in 1.6.1 which got released today.

closed 2021-05-10 by foosel