OctoPrint/OctoPrint #4120: Gcode layer number mismatch after restart of partial print

KenLucke, 2021-05-03:
### The problem

I accidentally interrupted a print, so I trimmed the code file(leaving the original layer numbers intact), reloaded it and started it from the point at which I disturbed it.  The “State” panel picked up the correct layer number, as did the Dashboard plugin layer indicator, but the gcode viewer started recounting from line 1. 

IOW, the first layer I restored at was layer 301 in the original file.  I cut everything out between layers 0 and 301, reset the extruder to what the last G1 E[x] was (at the end of layer 300), Homed, then raised to layer 301's Z height and restarted.  Dashboard and the “State” panel showed layer #301, code viewer showed layer #1.  10 layers later, Dashboard and State were correctly showing layer #310, gcode viewer was showing layer #10. , so it wasn't just initializing wrong and the later recovering.

Partial file used below:

[Main_Float_PLA_Partial [Blue].zip](https://github.com/OctoPrint/OctoPrint/files/6413023/Main_Float_PLA_Partial.Blue.zip)

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Print in process.

### Version of OctoPrint

1.6.0

### Operating system running OctoPrint

Mac OS 10.15.7

### Printer model & used firmware incl. version

N/A

### Browser and version of browser, operating system running browser

Firefox 8.7.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20210502225911.zip](https://github.com/OctoPrint/OctoPrint/files/6413027/octoprint-systeminfo-20210502225911.zip)

foosel, 2021-05-03:
The GCODE viewer and OctoPrint core do not parse (non standardized) slicer numbers to determine the real layer number, they just count layers contained in the file. AFAIK either Print Time Genius or DisplayLayerProgress however, which you have installed, DO parse slicer comments, so this is overriding the display in the state panel and also in your dashboard. So that's what you are seeing here - not a bug in OctoPrint, that's behaving as designed, but rather a inconsistency caused by some third party plugins that you have installed.

closed 2021-05-03 by foosel

KenLucke, 2021-05-03:
OK.  I  can live with it :).  Thanks for the explanation.