OctoPrint/OctoPrint #4126: G-Code script execution  on OctoPrint start

DoganM95, 2021-05-05:
### Is your feature request related to a problem? Please describe.

I want to e.g. level my bed every time OctoPrint boots by sending like 5 lines of gcode. 
Currently u have to do this by typing each command in the terminal on every boot of octoprint.

### Describe the solution you'd like

Instead, octoprint should add a "On system boot" to the category "G-Code Scripts" in the settings.

### Additional context

[image]

foosel, 2021-05-05:
"On system boot" doesn't really make sense. Just because OctoPrint is running out doesn't mean there's an actual printer connected.

The GCODE scripts already offer an "after connection" option which, considering that most printers reboot on serial connect, would make a lot of sense here. Could you explain what speaks against using *that*?

DoganM95, 2021-05-07:
Why doesn't it make sense? My printer is connected to the same AC-socket as my pi, so when my printer turns on, so does the pi and vice versa and i never plug off the usb connection between Printer mainboard and pi. Many people even go as far as powering the pi using the printers PSU so yes, for me and a lot of other people it indeed means just because octoprint is running it does mean a printer is connected.

Regarding the "After connection", I tried to use this as an alternative but this never fires for me unfortunately. Shouldn't this start, as soon as a client connects to the octoprint instance, e.g. using the web-UI? 

Anyway, what speaks against that is still the missing ability to run a gcode script as soon as octoprint has re-/booted. So when i turn on my printer and get a cup of coffee, in this case i could expect it to have leveled my bed when i am back.
In case of "after connection" i would have to wait for Octoprint to be ready to connect (webserver started, etc.), then get a device and connect it using an app/web and go get my coffee which is a lot less automation (coffee just as an example for a use case)

cp2004, 2021-05-07:
After connection is after a connection to the printer has been established, not the web UI or anything else, only what happens after the connection is established to the 3D printer.

Auto connect on startup + after connection script does exactly what you are looking for. If the after connect script does not run, then we would need a bug report for that to analyse and take a look.

Running a gcode script when OctoPrint has booted is impossible unless there is a printer connected. There's simply no way around this, there is nothing to receive what you send.

DoganM95, 2021-05-07:
Thanks for the clarification !
In that case, i will edit this issue to a bug-report and provide details.

foosel, 2021-05-07:
If you could also explain to me how you thought

[image]

meant "after some client has connected to OctoPrint's interface", that would help. Not trying to diss you here, I'm trying to understand how this wording could be misunderstood to mean what you thought it meant and how it can be improved to prevent other people from having the same misunderstanding in the future.

DoganM95, 2021-05-07:
This happened as there is quite some room for interpretation imo, as people mod their printers. 
I personally for example see "Printer" as a definition for the hole ecosystem that 3d prints, meaning a word that includes the physical printer itself, all attached hardware like the pi and all the software running on it, including octoprint.

This might sound comprehensible, but being unable to test it this is all i could have assumed.

Now as you explained, that a serial connection between octoprint and the printers mainboard is meant, it totally makes sense, seeing it from the perspective of octoprints software itself. 

So it could help captioning "after connection" as a serial connection and nothing else

closed 2021-05-07 by DoganM95