OctoPrint/OctoPrint #4127: Serial mistmatch error on Ender 3 when

inigoml, 2021-05-06:
### The problem

Easy to reproduce:
1. Set bed temperature by hand at Ender 3 display
2. Connect from Octoprint.

However, if you have not interacted by hand with Ender 3 display, no problems.
Workaroud: always connnect Octoprint after turning on printer and before manually sending any command from display.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.0

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Ender 3, Marlin 2.0.8-bugfix

### Browser and version of browser, operating system running browser

Firefox 88

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

This problem is related with Preheat or Nozzle functionality in Marlin.
If previously I've commanded to preheat bed manually via user display, as soon as I connect Octoprint to Marlin this error is displayed and also a reset requirement shown in Ender 3 display.
However, if no command has been introduced from Ender 3 console, no problem.

[octoprint-systeminfo-20210506152030.zip](https://github.com/OctoPrint/OctoPrint/files/6434769/octoprint-systeminfo-20210506152030.zip)
[image]

jneilliii, 2021-05-06:
Have you stopped [backpowering](https://community.octoprint.org/t/printers-known-to-have-backpowering-issues/25913) on your USB line?

inigoml, 2021-05-06:
> Have you stopped [backpowering](https://community.octoprint.org/t/printers-known-to-have-backpowering-issues/25913) on your USB line?

Yes. As soon as Ender is shutdown, motherboard stop working. Weeks ago wasn't, but right now is not powering motherboard.

This behavior was also present in v1.5.2 dockerized in a Linux x64 box and now in Octopi. And only happens when connecting and a command has been previously sent via display wheel (bed preheat, for example).

inigoml, 2021-05-06:
Motherboard 32bit, v4.2.2

foosel, 2021-05-06:
First of all:

```
 env.plugins.pi_support.throttle_state: 0x80000
```

Your pi isn't running stable, it saw a soft overheat issue since last boot. Improve cooling.

Then

```
env.plugins.pi_support.octopi_version: 1.0.0
```

I'm not aware of this OctoPi version, where did that come from?

Then, please enable serial.log, reproduce the issue with it enabled, then immediately create a few system bundle and share it.

In all likelihood, based on the error you are seeing, the issue is caused by *your firmware* and OctoPrint is just reacting to that.

inigoml, 2021-05-06:
Will provide serial.log by tomorrow.

This is official Octopi and executed upgrade to 1.6.0 from web frontend when offered update. Also some plugins installed. Running in a Raspi 3B+. Nothing special.

cp2004, 2021-05-06:
OctoPi 1.0.0 is what the current nightly development builds are showing up as, not the official stable release (which is 0.18). 

Definitely seems a lot more likely it is a firmware bug to me, but I guess we won't know until we get the serial.log to look at.

inigoml, 2021-05-07:
Some comments:
- This behavior was also present in 1.5.2 dockerized (from a debian9 server). So, I think it's not related with OctoPi. 
- My Octopi is the 0.18.0, updated from command line with "apt get update && apt get upgrade" to get latest packets and for security reasons.
- Octoprint was 1.5.2, but upgraded automatically from users interface when 1.6.0 was offered.

Attached extra info with serial.log.
First I powered on ender 3. No issues
Later I was changing some parameters (bed temperature, etc). No issues
Finally I disconnected USB and reconnected and problem arised.

So, It appear not to be an issue with the previous sent commands to Marlin but some kind of serial issue when disconnecting and reconnecting.

[octoprint-systeminfo-20210507112402.zip](https://github.com/OctoPrint/OctoPrint/files/6440576/octoprint-systeminfo-20210507112402.zip)
[serial.log](https://github.com/OctoPrint/OctoPrint/files/6440578/serial.log)

foosel, 2021-05-07:
As far as I can see from the Marlin code, this error should only happen when you have multiple serial ports configured. Is this the case here? Please try with a current stable firmware version, we've seen a ton of serial stability issues with the bugfix branch (which is actually a development branch) over the past few months. Considering that it happens across OctoPrint versions and across OSes, this is more and more likely to be a firmware issue than anything else.

>     * My Octopi is the 0.18.0, updated from command line with "apt get update && apt get upgrade" to get latest packets and for security reasons.

Whatever you downloaded, it wasn't 0.18 but a development nightly build. The version number of OctoPi doesn't change by you applying system updates.

inigoml, 2021-05-07:
Ok. Thank you. Will try with stable build as soon as I get bltouch to work with 4.2.2 board. Official firmware is a mess. :)
Anyway, I can live with this issue. It's only a matter of restarting ender 3 before connecting to OctoPrint.

cp2004, 2021-05-08:
@inigoml Is this [Jyers](https://github.com/Jyers/Marlin) firmware? Just seen another user with this issue, if it is from the same firmware then it is likely a bug there I think.

inigoml, 2021-05-09:
Not really. I'm using this daily builds. Started with April 22nd and now with May 7th.
https://marlin.crc.id.au/

Anyway this only happens when connecting for the first time from Octoprint. Once restarted Ender and (important) unplugging and plugging (not always needed but sometimes it's necessary) and established connection, there are no issues. And if I let octopi running and connected to Ender 3, there is no problem when powering Ender 3.

I agree that this is probably a firmware problem related with some kind of buffering at USB/Serial port.

foosel, 2021-05-10:
Since this is currently not actionable for me, I'll close this so it doesn't clutter the bug tracker. Comments are still possible though.

closed 2021-05-10 by foosel

inigoml, 2021-05-24:
Only for information:
- With meatpack enabled in Marlin Firmware Service from Creality, at 128000 bps, no problems.
- With Marlin 2.0.8 (official) with meatpack enabled and 250000 bps, no problems.