OctoPrint/OctoPrint #4141: GCodeViewer/Layer Info: add the estimated time a specific layer will be printed

felinh, 2021-05-19:
### Is your feature request related to a problem? Please describe.

First of all, congrat and many thanks for this wonderful tool.

This request is motivated by the need to monitor some critical phases of a print (nut insersion, bridge junction, 1st layer of multi part printing). Goal is to be able to be present near to the printer for specific layers and thus to know (approx.) when they will be printed.

### Describe the solution you'd like

while browsing the layers in the GCodeViewer/Layer Info, it would be nice to see _when_ the layer will be printed (today there is only the duration)

cp2004, 2021-05-19:
I believe TimeToFilament is what you want: 

https://plugins.octoprint.org/plugins/TimeToFilament/

I know this is not a filament change, but it claims to be extendable for many other things and using regular expressions you could try to make it do what you want it to.

felinh, 2021-05-20:
Thank you Charlie, this plugin will do the job!

closed 2021-05-20 by felinh