OctoPrint/OctoPrint #4142: [Request] Process heater power reported by some firmware via M105

matyay, 2021-05-19:
### Is your feature request related to a problem? Please describe.

I've noticed that OctoPrint does not parse heater power even if it is reported by a printer via M105  (I've been through the code). I'd like this data to be processed and exposed via the server API. Heater power may not be the most important thing to know for an end user but it is very informative when determining limits of one's hotend.

### Describe the solution you'd like

At least I'd like OctoPrint to parse reported heater power and expose it via the server API. It would be nice to be able to see the value in the "Temperature" tab next to actual and target temperatures.

The RepRap wiki gives an example of a M105 output with heater power data: https://reprap.org/wiki/G-code#M105:_Get_Extruder_Temperature

foosel, 2021-06-22:
I'm aware of the existence of the heater power fields, however to this day I have not been able to find documentation as to *how* this stuff is supposed to be parsed. What are the units? What's the precise format? Is it consistent across firmware variants out there (pretty sure that not because *nothing* is consistent, but maybe there's at least *some" agreement on something there...)?

If you or anyone else can provide answers to the above, I can look into implementing the parsing side of things. I frankly would prefer not to have to do yet another firmware source code reading of various code bases out there just to figure this out, and there's currently nothing of substance on those fields in the reprap wiki.

matyay, 2021-06-23:
Hmmm, so the most complete information about M105 response is under the link I've posted in the description. As far as I know it is indeed very firmware specific.

Anyways, maybe it could be possible at least to allow a plugin to directly access the response string? This way one could have a firmware-specific plugin while the core OctoPrint code would be kept generic.

jneilliii, 2021-06-23:
This could be achieved the same way my single file plugin for graphing fan speed and the [plotly temp graph](https://github.com/jneilliii/OctoPrint-PlotlyTempGraph) plugin I made. 

Here's the link to the single file plugin as example.

https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/blob/master/test_plotly_graph_fan_speed.py

foosel, 2022-06-23:
> so the most complete information about M105 response is under the link I've posted in the description

[And guess who wrote a lot of that based on her experiences while developing OctoPrint](https://reprap.org/mediawiki/index.php?title=G-code&diff=185176&oldid=185175). Alas, no one ever bothered to document the fields anywhere, and just mentioned their existence on the wiki node as well after I tried my best to make it an actual knowledge source. It's a mess as always and so this request will have to remain unfulfilled until we finally get some clarity in the 3d printing community on what these fields actually express in which unit beyond a generalized "heater power", and whether that is identical across at least the most common firmware variants.

Fl0Cri, 2023-03-15:
I landed here while trying to search for information about what these values should be.

I've done some quick source code digging in a few firmwares. If it can help, here are my findings.

(Note: I've not tried on working printers, it's just some code interpretation so I may miss something, especially on Prusa/Marlin)

### Klipper
**No support**, only temperature in format `Tn:XXX /YYY B:XXX /YYY`: [klippy/extras/heaters.py#L309-L317](https://github.com/Klipper3d/klipper/blob/c179db3d4331db9d85c7acfdaa1e96e295d277ba/klippy/extras/heaters.py#L309-L317)

### RepRapFirmware
**No support**, it has exactly the same `printf` style formatting string as Klipper. According to the comments, the format seems to have been agreed here in #2590:
- Beds and chambers: [src/GCodes/GCodes.cpp#L4498-L4533](https://github.com/Duet3D/RepRapFirmware/blob/e56a9779fe8c9ce56f0e63a2de8977e249024b2d/src/GCodes/GCodes.cpp#L4498-L4533)
- Tools: [src/Platform/RepRap.cpp#L1261-L1300](https://github.com/Duet3D/RepRapFirmware/blob/3d0c990ab154c226af981713eb9c001f9a88cb46/src/Platform/RepRap.cpp#L1261-L1300)

### Marlin
**Supported**, the value seems to be an integer between 0 and 127 reflecting the pwm duty cycle of the heater:
- Where it is printed: [src/module/temperature.cpp#L4025](https://github.com/MarlinFirmware/Marlin/blob/bc450b45824140d9c0e73e6b04c427ca81de5b57/Marlin/src/module/temperature.cpp#L4025)
- `getHeaterPower()` returns a `soft_pwm_amount` variable: [src/module/temperature.cpp#L1188](https://github.com/MarlinFirmware/Marlin/blob/bc450b45824140d9c0e73e6b04c427ca81de5b57/Marlin/src/module/temperature.cpp#L1188)
- `soft_pwm_amount` is assigned here with the result of `get_pid_output_hotend()` divided by 2: [src/module/temperature.cpp#L1612](https://github.com/MarlinFirmware/Marlin/blob/bc450b45824140d9c0e73e6b04c427ca81de5b57/Marlin/src/module/temperature.cpp#L1612)
- Before being returned, the value in `get_pid_output_hotend()` is constrained between 0 and `MPC_MAX`: [src/module/temperature.cpp#L1547](https://github.com/MarlinFirmware/Marlin/blob/bc450b45824140d9c0e73e6b04c427ca81de5b57/Marlin/src/module/temperature.cpp#L1547)
- According to the config file, `MPC_MAX` is in the range 0..255: [Configuration.h#L697](https://github.com/MarlinFirmware/Marlin/blob/10983d0cfe3e353ba465cfcadf8aeb51327f1197/Marlin/Configuration.h#L697)

### Prusa-Firmware
**Supported**. Quicker search here, but the `soft_pwm` variable (kind of equivalent to `soft_pwm_amount` in Marlin) seems to always be assigned with a `>> 1` (equivalent to a division by 2), like here: [Firmware/temperature.cpp#L1957](https://github.com/prusa3d/Prusa-Firmware/blob/16ec3e376f1155d0ffa119b6514ee373fc9db038/Firmware/temperature.cpp#L1957). This should lead to a number between 0 and 127

### Repetier
**Supported**. It should be an integer between 0 and 255. There's a note about it in their changelog: [changelog.txt#L191](https://github.com/repetier/Repetier-Firmware/blob/2bbda51eb6407faf29a09987fd635c86818d32db/changelog.txt#L191)

---
So in conclusion all firmwares that report heater power are sending an integer to represent a "percentage" of the heater capacity. But unfortunately the range seems to differ between Marlin/Prusa (0..127) and Repetier (0..255).

The 0..127 range seems odd, as most other gcodes documented in the RepRap wiki use ranges between 0 and 255. Maybe someone with a Prusa could have a try and report if it's correct ?