OctoPrint/OctoPrint #4143: [Request] Merge settings overlay with already configured settings

jneilliii, 2021-05-20:
### Is your feature request related to a problem? Please describe.

When developing OctoDash Companion plugin I added a `__plugin_settings_overlay__` to add system menu items for restarting the OctoDash related services. What I found was that if any setting already existed in the system section it would prevent the overlay from happening, even though my settings didn't conflict with any pre-existing setting.

### Describe the solution you'd like

Would like for `__plugin_settings_overlay__` to be merged with existing settings recursively so that as long as the specific setting isn't already in config.yaml that it would be applied, not just the first level of the hierarchy. See code example below.

### Describe alternatives you've considered

During init forcefully adding these settings to config.yaml, which would remain on the system if the plugin was uninstalled, kind of blah about that. 

### Additional context

Example code:
```
__plugin_settings_overlay__ = {
    'system': {
        'actions': [{
                'action': 'octodash_start',
                'command': 'sudo service getty@tty1 start',
                'name': 'Start OctoDash'
            }, {
                'action': 'octodash_stop',
                'command': 'sudo service getty@tty1 stop',
                'name': 'Stop OctoDash',
                'confirm': 'You are about to shutdown OctoDash.'
            }, {
                'action': 'octodash_restart',
                'command': 'sudo service getty@tty1 restart',
                'name': 'Restart OctoDash',
                'confirm': 'You are about to restart OctoDash.'
            }
        ]
    }
}
```

jneilliii, 2021-06-15:
Hook makes sense too.

foosel, 2021-06-15:
I feel like in a cooking show where I pull a finished dish from under the counter and go "I already prepared that here"...

See above commit, new hook `octoprint.system.additional_commands`. Will be in 1.7.0.

cp2004, 2021-10-11:
1.7.0 has been released 🎉