OctoPrint/OctoPrint #4144: Hide video / stop streaming when Control tab is not active

geerlingguy, 2021-05-20:
### Is your feature request related to a problem? Please describe.

A few times, I've noticed that if I leave a tab open on OctoPrint to monitor the status on the GCode Viewer (or another one of the main tabs), but I previously had clicked on the Control tab (which displayed the webcam stream output), my computer's network connection is still constantly streaming the full video feed from the OctoPrint server.

In my case, it's not a huge amount of data (1-2 MB/sec constantly from a Raspberry Pi). But it is significant, and if nothing else just adds a bit of extra WiFi traffic that's being delivered to my computer constantly.

### Describe the solution you'd like

It would be nice if when the Control tab was not actively selected, the stream would stop. Then when the user clicks on the Control tab, the stream starts up again.

Ideally, there would be some sort of timeout so if you are clicking between tabs quickly, it wouldn't keep reloading the stream (unless that's a relatively easy operation that won't cause other issues), so the stream would be cut off if not visible after 30s or so.

### Describe alternatives you've considered

If I just go to the tab I want (e.g. GCode Viewer) and refresh the page, then it stays down at 2-4 KB/sec instead of 1-2 MB/sec.

Alternatively, I could completely disable the webcam preview (but that's no fun!).

### Additional context

Screenshot of the situation I'm describing—I have the GCode Viewer tab open but had earlier opened the Control tab. You can see the data usage via Activity Monitor (note there is only one other tab open in Safari currently, and it's this 'New Issue' tab—the traffic is all coming through the OctoPrint tab):

[image]

cp2004, 2021-05-20:
So this is actually already implemented, *however* not for Safari. This comment suggests there was a bug trying to reload the stream:

https://github.com/OctoPrint/OctoPrint/blob/884ccee5f8ff8b5f040cb265ba40bb9f55bd7a89/src/octoprint/static/js/app/viewmodels/control.js#L684

Maybe it still exists, maybe not. Would have to go through BrowserStack to test it.

foosel, 2021-05-20:
Just did, the bug is still there.

Other browsers will stop the stream once you unset the embedding img's `src`. Safari still doesn't, so there's no way to unload the webcam stream in that browser. Workaround has to stay.

@geerlingguy if you are using Safari I suggest you use a different browser to access OctoPrint if the webcam traffic is a concern, or switch to HLS based streaming (I *think* that should properly unload, but that's code I'm admittedly not as familiar with since I didn't write it myself).

If you are *not* using Safari then this is actually a bug report because the functionality should already be implemented exactly as requested in other browsers. So... what browser are you using? :)

closed 2021-05-20 by foosel

geerlingguy, 2021-05-20:
@foosel - Heh, I am using Safari, but good to know. Ah well... I'll try HLS and see what happens.