OctoPrint/OctoPrint #4146: USB MK3S+ FW 3.10.0 - Resume from gcode pause fails to resume with nozzle temperature control

mjroux1984, 2021-05-20:
### The problem

When using gcode with embedded pause (M601) using Prusaslicer 2.3.1 (pause at specific layer height), the printer pauses correctly, but when resuming from either the printer or the Octoprint GUI, it will  resume printing with the nozzle temperature control off (will try to print with nozzle at room temperature).  The same gcode works correctly when run from the SD card on firmware 3.10.0.  When the printer is downgraded to 3.9.3, the same gcode works correctly through octorprint USB as well as SD card.

The uploaded zip file is an example with pause after the first layer.
[pause_test_0.2mm_PETG_MK3SMMU2S_33m.zip](https://github.com/OctoPrint/OctoPrint/files/6517226/pause_test_0.2mm_PETG_MK3SMMU2S_33m.zip)

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

Octopi 0.18.0

### Printer model & used firmware incl. version

Prusa MK3S+ with MMU2s updgrade, firmware 3.10.0

### Browser and version of browser, operating system running browser

Google Chrome Version 90.0.4430.212 (Official Build) (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20210520165526.zip](https://github.com/OctoPrint/OctoPrint/files/6517242/octoprint-systeminfo-20210520165526.zip)

[pause_test_0.2mm_PETG_MK3SMMU2S_33m.zip](https://github.com/OctoPrint/OctoPrint/files/6517248/pause_test_0.2mm_PETG_MK3SMMU2S_33m.zip)

cp2004, 2021-05-20:
Could you enable and the serial.log, reproduce the issue then upload this file. Do you have any gcode scripts set for pause and resume in OctoPrint? Can't really tell what the issue might be, but looking at the serial.log would help massively.

mjroux1984, 2021-05-20:
> Could you enable and the serial.log, reproduce the issue then upload this file. Do you have any gcode scripts set for pause and resume in OctoPrint? Can't really tell what the issue might be, but looking at the serial.log would help massively.

I can, I'll just need some time to accomplish.  Thanks for quick reply.

closed 2021-05-20 by mjroux1984

foosel, 2021-05-20:
If it works with the old firmware version and fails with the new one, the problem is most likely with the firmware, not OctoPrint.

mjroux1984, 2021-05-20:
> Could you enable and the serial.log, reproduce the issue then upload this file. Do you have any gcode scripts set for pause and resume in OctoPrint? Can't really tell what the issue might be, but looking at the serial.log would help massively.

I attached serial.log to main text.

cp2004, 2021-05-20:
Yeah, definitely more like a firmware/configuration issue. The temperature starts to go down again on it's own:
```
2021-05-20 17:35:58,172 - Send: N337 M601*35
2021-05-20 17:35:58,899 - Recv: ok
2021-05-20 17:35:58,901 - Send: N338 G1 Z0.600*114
2021-05-20 17:35:58,903 - Recv: echo:enqueing "G1 E-1.000 F2700"
2021-05-20 17:35:58,932 - Recv: // action:paused
2021-05-20 17:35:58,932 - Printer signalled that it paused, switching state...
2021-05-20 17:35:58,932 - Changing monitoring state from "Printing" to "Pausing"
2021-05-20 17:35:58,952 - Recv: ok
2021-05-20 17:35:58,954 - Changing monitoring state from "Pausing" to "Paused"
2021-05-20 17:35:59,358 - Recv: LCD status changed
2021-05-20 17:36:00,104 - Recv: T:230.6 /0.0 B:85.1 /90.0 T0:230.6 /0.0 @:0 B@:127 P:38.5 A:31.5
2021-05-20 17:36:02,200 - Recv: T:229.9 /0.0 B:85.1 /90.0 T0:229.9 /0.0 @:0 B@:127 P:38.6 A:31.7
2021-05-20 17:36:04,297 - Recv: T:229.9 /0.0 B:85.1 /90.0 T0:229.9 /0.0 @:0 B@:127 P:38.6 A:31.4
```

mjroux1984, 2021-05-20:
> If it works with the old firmware version and fails with the new one, the problem is most likely with the firmware, not OctoPrint.

I agree, but I figured I'd create this just in case.  The 3.10.0 firmware changelog mentions "optimizations for print hosts" which may be related.  Complete list of changes shown here: [https://www.prusa3d.com/drivers/firmware-changelog/]

mjroux1984, 2021-05-25:
For future reference, it appears adding gcode script commands for "after pause" and "before resume" will resolve this issue for 3.10.0, please reference comment below:
https://github.com/prusa3d/Prusa-Firmware/issues/3148#issuecomment-843442610