OctoPrint/OctoPrint #4147: Nozzle crashes past 0,0 keeps pushing, extruder pukes filament

Jasonpenman, 2021-05-20:
### The problem

Randomly on starting a print it goes through the heat up fine, bed level runs, but instead of the purge line it goes to 0 0 and keeps going, the extruder runs at the same time firing out a heap of filament. Sounds horrible. 3 times in the last 2 days I have had to run over and turn the printer off, never had this issue before. Same version of cura etc.

Happened 3 times in the last 2 days. Latest marlin2.0 bugfix.

After restarting the printer the same gcode will print fine

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

happens at random

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

octopi 0.18

### Printer model & used firmware incl. version

Ender 6 skr 2, recent marlin 2.0 bugfix

### Browser and version of browser, operating system running browser

Chrome latest version, any

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/90.0.4430.93 Safari/537.36 Edg/90.0.818.56
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: True
connectivity.enabled: True
connectivity.online: True
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: True
env.hardware.cores: 4
env.hardware.freq: 1500.0
env.hardware.ram: 3958255616
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: True
octoprint.safe_mode: False
octoprint.version: 1.6.1
systeminfo.generator: zipapi

DanErshen, 2021-05-20:
I have the same issue on my 5+ running Marlin’s release from February sliced in Prusa slicer. 

So I don’t think it’s firmware related nor slicer related. I can only speak for myself, but this had never happened to me prior to the most recent OctoPi update.

mattborkin, 2021-05-20:
Same thing happens to me. Sometimes it happens once a day, sometimes it happens three or four times back to back. I’m using the Marlins firmware from February and this happens across multiple slicers. If it’s any help, the filament extrudes out an extremely high rate.  It’s always the same corner. None of this happened until two octopi updates ago.

cp2004, 2021-05-20:
There are no logs attached to this ticket, please add them. You can download a system info bundle from OctoPrint which includes all the information you need, instead of hunting around trying to find it.

We also need someone to test it in safe mode - just because it happens at random, does not mean you cannot test safe mode. You need to run safe mode until you are confident that this issue no longer happens.

If there is anyone else who would like to comment here, we also need the logs from *everyone*. Ideally, with **serial logging enabled** and a log captured when you find this issue.

---

My initial opinion - it's probably an issue with relative vs. absolute extrusion, and you need to adjust some settings to fix this.

mattborkin, 2021-05-20:
Thanks, I’ll work on a log tomorrow. It happened with every file, there’s no specific file that causes. The same file will eventual work with no changes being made. 

I have tried with absolute and relative extrusion. Happens regardless.

cp2004, 2021-05-20:
Upload one of your gcode files anyway so we can take a look, we need as much information as possible. Multiple people going 'I have this issue' is not very helpful.

mattborkin, 2021-05-20:
> Upload one of your gcode files anyway so we can take a look, we need as much information as possible. Multiple people going 'I have this issue' is not very helpful.
Oh, for sure. I’m mobile at the moment, and can try tomorrow. Thanks.

cp2004, 2021-05-27:
Does anyone have any information to share here? If we don't receive a response in another 7 days we will have to close this issue as  there is nothing we can do without information to solve it.

cp2004, 2021-06-05:
Since no further information was shared, there is nothing we can do about this bug report.

If this issue still persists, please provide the needed information and request a reopen.

closed 2021-06-05 by cp2004