OctoPrint/OctoPrint #4151: GCode viewer's origin is not updated when print origin changed

BluSpanner, 2021-05-30:
### The problem

1. Settings
2. Printer profile
3. Edit
4. Print bed & build volume
5. Origin: Center (was Lower Left)
6. Save
7. GCode Viewer
8. [the origin is is incorrect] 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.6.1 Python 3.7.3 OctoPi 0.18.0

### Operating system running OctoPrint

Windows 10

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20210530050625.zip](https://github.com/OctoPrint/OctoPrint/files/6565671/octoprint-systeminfo-20210530050625.zip)

cp2004, 2021-05-30:
I'm not able to reproduce this - as of 1.6.1 the gcode viewer should respond to printer profile changes as far as I know. So you can see what I'm doing and make sure we are doing the same thing, you can watch:
[image]

Does it change after a page reload? Are we looking at the same thing?

cp2004, 2021-06-05:
Hey @BluSpanner are you able to confirm the above questions?

> Does it change after a page reload? Are we looking at the same thing?

cp2004, 2021-06-15:
As I'm unable to reproduce the problem and have not received a reply from @BluSpanner this will be closed. Feel free to request a reopen if you can provide more details on the issue.

closed 2021-06-15 by cp2004