OctoPrint/OctoPrint #4152: Managing files in the filemanager randomly causes web side of octoprint to stop functioning.

credomane, 2021-05-31:
### The problem

I think issue #4119 might still be floating around?

Ever since updating to 1.6.1 from 1.5.x (somehow skipped 1.6.0) I've noticed that managing files in the file manager causes the http side of octoprint to stop functioning completely at random. Mostly when I delete a file doesn't seem to matter if I have a print running or not. I've had it happen either way. Octoprint seems to completely ignore http side of things while (if I have one running when it happens) the print job from octoprint over the serial connection continues to function without issues.

Having found issue 4119 I'll check back after the 10000s timeout, (what is that about 3hours?), and see if the webpage lives again. Normally I just restart octoprint immediately or after the print finishes without even checking the webpage again. 

I haven't gotten it to happen in safemode yet. Keep switching out of safe-mode for ArcWelder plugin and forgetting to switch back after it runs. I'll switch to safemode (once this print finishes and to see if webpage lives again after 3 hours) and just live without ArcWelder for a while.

I also do not have a systeminfo bundle to provide atm since the webpage is currently non-functional. Once tomorrow morning rolls around I'll check if the webpage is functional, restart octoprint if not, and update this ticket additional info plus systeminfo bundle at that time.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

I'm a failure that forgets to switch to safe mode before messing about in filemanager. I *will* eventually remember.

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

Rpi3 w/ OctoPi 0.17.0

### Printer model & used firmware incl. version

Ender 3. Marlin 2.0.7.2

### Browser and version of browser, operating system running browser

Firefox 88.0.1 on Manjaro Linux

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

credomane, 2021-05-31:
Well the print finished, successfully, about 4 hours ago and the webservice side of octoprint was still non-functional until I rebooted the pi a few minutes ago. I've switched to safe-mode and will see if I can make the problem re-occur over the next week or so.

cp2004, 2021-05-31:
2nd user on Discord reporting this issue today, with deleting a file. [link to discussion](https://discord.com/channels/704958479194128507/705046975669600298/848975155598196756)

Things to note from the conversation:
* I've not reproduced it on my environment, and it happens at random. Same file (different names) sometimes freezes on the 1st, sometimes 4th.
* There is nothing generated in the log file at the time of pressing delete, either here or from discord (I haven't seen the discord one, but so I am told)
* The request is sent to the server, but never answered. See screenshot:
  [image]

Debug logs on their way to see if there is anything to show *where* it may be hanging.

credomane, 2021-06-01:
Closing this. Definitely seems to be a bug in PrintTimeGenius. Seems limited to PrintTimeGenius on python 3.

closed 2021-06-01 by credomane