OctoPrint/OctoPrint #4154: Model size detection should be disabled for belt printers (Z=0)

deftdawg, 2021-06-02:
### The problem

This beta feature:  `Enable model size detection and warn if model exceeds print bed` popups up warnings if the bed depth is set to `0`...  I set the depth to `0` because I have a belt printer with infinite depth

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/90.0.4430.212 Safari/537.36 connectivity.connection_check: 1.1.1.1:53 connectivity.connection_ok: true connectivity.enabled: true connectivity.online: true connectivity.resolution_check: octoprint.org connectivity.resolution_ok: true env.hardware.cores: 4 env.hardware.freq: 900 env.hardware.ram: 915066880 env.os.bits: 32 env.os.id: linux env.os.platform: linux env.plugins.pi_support.model: Raspberry Pi 2 Model B Rev 1.1 env.plugins.pi_support.octopi_version: 0.17.0 env.plugins.pi_support.throttle_state: 0x50000 env.python.pip: 21.1.1 env.python.version: 3.7.3 env.python.virtualenv: true octoprint.safe_mode: false octoprint.version: 1.6.1 printer.firmware: Marlin 2.0.8.1 (May 19 2021 15:12:21) systeminfo.generator: systemapi

### Printer model & used firmware incl. version

Creality CR-3D

### Browser and version of browser, operating system running browser

Chrome / Windows

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20210602062207.zip](https://github.com/OctoPrint/OctoPrint/files/6583709/octoprint-systeminfo-20210602062207.zip)

cp2004, 2021-06-02:
You can disable the feature yourself, that's why there's a switch. Setting it to 0 does not seem conclusive enough to know 'this is a belt printer' and automatically ignore it - it would highlight incorrect configuration quickly.

On the other hand, a switch in the printer profile/bed size stuff that said 'this is a belt printer' would be a better option, since that *would* be conclusive.

foosel, 2021-06-14:
This isn't just an improvement, this is a full blown feature request. Belt style printers don't necessarily have to have the z axis be infinite, it could also be x or y, depends entirely on the belt printer in question and the way the files are sliced. Not only the model size detection, the GCODE viewer would also need adjustment in some way. Possibly more places. This is non trivial.

alphanumericaltycoon, 2021-12-29:
I'd also welcome this feature

cp2004, 2022-04-29:
Work on a PR yourself, or in the past we've had people sponsor FRs for someone in the community to work on.