OctoPrint/OctoPrint #4156: [Request]  UPS integration / power fail recovery

mytechguyri, 2021-06-03:
### Is your feature request related to a problem? Please describe.

Like it or not, power failures happen to everyone eventually, while having a UPS can help mitigate this to a certain extent, there needs to be more intelligence on the part of Octoprint as to how it handles such failures.

### Describe the solution you'd like

I would like to see a feature in Octoprint that when a UPS notifies the host that a power failure has occured (either via serial interface or GPIO), Octoprint would immediately pause the current print, save the current print state, and perform an orderly shutdown of the host.

Upon restoration of power, Octoprint would "see" that there was an interrupted print saved state file, and ask the user if they would like to resume the print.

Unlike the Marlin power failure resume feature, that writes to the SD card for every layer, this would leverage the fact that a UPS is connected, and only write the print state to the SD card when a failure is reported by the UPS.    This would allow a simple UPS such as those that use 2 18650 cells to power the PI just long enough for the orderly shutdown of the system, without needing to worry about keeping the Pi and printer powered throughout the entire power failur, which of course would require a much more substantial UPS.  

### Describe alternatives you've considered

Big UPS with lots of mAh to power through any power failure.

cp2004, 2021-06-03:
See also #3573 - there was a plugin made, not sure if it works. This is still a great plugin candidate, since it does not benefit most users and can all be achieved with an OctoPrint plugin, created by someone that has the hardware/capability to do this.

jneilliii, 2021-06-03:
Shouldn't this be considered duplicate of linked issue and closed?

cp2004, 2021-06-04:
As you wish.

Duplicate of #3573

closed 2021-06-04 by cp2004