OctoPrint/OctoPrint #4157: GcodeInterpreter does not calculate arc moves

ldursw, 2021-06-04:
### The problem

When fixing an issue with arc moves not being correctly calculated in PR OllisGit/OctoPrint-SpoolManager#182 it was pointed out in a comment that the analysis code was copied from OctoPrint and the original file also has the same issue.

https://github.com/OctoPrint/OctoPrint/blob/8db57f72ab90c7ed9ef1fa5a92d11f8ff904d40d/src/octoprint/util/gcodeInterpreter.py#L420

The plugin only cares about the extrusion length so adding `"G2", "G3"` is enough. OctoPrint also calculates the XYZ moves so it would require some special handling for `IJ` parameters.

```
# Original file without arcs
$ OctoPrint/bin/octoprint analysis gcode Quest2ControllerCoverLeft.gcode
DONE:19.82067164313048s
RESULTS:
dimensions:
  depth: 180.0
  height: 32.4
  width: 156.883
extrusion_length:
- 6933.180119999942
extrusion_volume:
- 0.0
printing_area:
  maxX: 158.983
  maxY: 200.0
  maxZ: 32.4
  minX: 2.1
  minY: 20.0
  minZ: 0.0
total_time: 112.9555746417688

# File with arcs without G2/G3 added to line 420
$ OctoPrint/bin/octoprint analysis gcode Quest2ControllerCoverLeft.aw.gcode
DONE:2.820354064926505s
RESULTS:
dimensions:
  depth: 180.0
  height: 32.4
  width: 156.372
extrusion_length:
- 374.4529999999963
extrusion_volume:
- 0.0
printing_area:
  maxX: 158.472
  maxY: 200.0
  maxZ: 32.4
  minX: 2.1
  minY: 20.0
  minZ: 0.0
total_time: 41.78244642603364

# File with arcs with G2/G3 added to line 420
$ OctoPrint/bin/octoprint analysis gcode Quest2ControllerCoverLeft.aw.gcode
DONE:3.956256238743663s
RESULTS:
dimensions:
  depth: 180.0
  height: 32.4
  width: 156.857
extrusion_length:
- 6933.180120000003
extrusion_volume:
- 0.0
printing_area:
  maxX: 158.957
  maxY: 200.0
  maxZ: 32.4
  minX: 2.1
  minY: 20.0
  minZ: 0.0
total_time: 112.24193040348119
```

Adding the patch from the PR fixes `extrusion_length` and `maxX` is close but not quite right. It uses the XYZ values from G2/G3 but doesn't consider the radius of the arc in the travel.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

Linux

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20210604220421.zip](https://github.com/OctoPrint/OctoPrint/files/6601423/octoprint-systeminfo-20210604220421.zip)
[Quest2ControllerCoverLeft.gcode.txt](https://github.com/OctoPrint/OctoPrint/files/6601426/Quest2ControllerCoverLeft.gcode.txt)
[Quest2ControllerCoverLeft.aw.gcode.txt](https://github.com/OctoPrint/OctoPrint/files/6601427/Quest2ControllerCoverLeft.aw.gcode.txt)

cp2004, 2021-06-04:
Yes, it doesn't take into account arc commands. It's on the list to fix (see some of the later comments in #4086, which was similar issue but in the frontend), and I'm hoping someone will be able to find some code they have already written here to work maybe from ArcWelder or the gcode viewer (again, see #4086)

The code used in the gcode viewer may not be the most efficient to translate directly to Python, but it does work.

Not sure what to label this with so we will get both 🙂

foosel, 2021-06-08:
This will be part of 1.7.0 thanks to the provided PR that just was merged 👍

closed 2021-06-08 by ldursw