OctoPrint/OctoPrint #4158: Dangerous temp offset +200° after reset!

rafpigna, 2021-06-05:
### The problem

Hello
I dont know what is happening since is my first days on octoprint, but every time I power on the board and the printer and log in octoprint, I found a temp offset of +200 celsius! This means that when I started a printing job with temp setted up at 195° for PLA, the printer setted the extruder temp to 395° ! Luckly I was looking to all the stuff on the screen and I immediately stop the print, but the extruder already reached 270° ! I tried to delete the offset and started the print again and all was fine. Then today I powered on again octoprint and the printer, started another file print (different from the first one) and again the +200° offset was there! 

I looked in the setting and I didnt found anything to set the offset to 0, so seems that everytime I restart the octoprint server, the +200 offset is there. 

Is this a bug/error or it's the expected behaviour? How can I avoid this or change the setting? 
Thanks

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Longer LK4 Pro with Marlin 1.1.9

### Browser and version of browser, operating system running browser

Chrome (latest) on Win10

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.77 Safari/537.36
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1200
env.hardware.ram: 915722240
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 3 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.6.1
printer.firmware: Marlin 1.1.9 (Github)
systeminfo.generator: systemapi

jneilliii, 2021-06-05:
This is not default behavior. Please provide requested systeminfo bundle.

rafpigna, 2021-06-05:
> This is not default behavior. Please provide requested systeminfo bundle.

Here the file. Thanks

[octoprint-systeminfo-20210605043149.zip](https://github.com/OctoPrint/OctoPrint/files/6601955/octoprint-systeminfo-20210605043149.zip)

cp2004, 2021-06-05:
Please ensure that the offset you are setting is OctoPrint's offset, **not** the offset for the filament manager plugin.

The temperature offsets in OctoPrint are here, and the maximum value is 50:
[image][image]

The plugin 'Filament Manager' has offset capabilities, and sometimes this is misconfigured as the material temperature. Please test this in safe mode or with filament manager disabled. If the issue persists, please upload the logs from the safe mode test.

cp2004, 2021-06-09:
@rafpigna have you been able to look at if it is the Filament manager plugin?

rafpigna, 2021-06-09:
> @rafpigna have you been able to look at if it is the Filament manager plugin?

Hello
Sorry for late reply
I can confirm that the offset was caused from the plugin (and make no sense for me that they install the plugin with a default 200 C degree, but I will write to the plugin developer). Thanks for your support.

closed 2021-06-09 by cp2004