OctoPrint/OctoPrint #4160: [Docs] Printer Profiles endpoint doesn't return list as mentioned

cp2004, 2021-06-07:
### The problem

Requesting `/api/printerprofiles`, I receive this:
```json
{
  "profiles": {
    "_default": {
      "axes": {
        "e": {
          "inverted": false, 
          "speed": 300
        }, 
        "x": {
          "inverted": false, 
          "speed": 6000
        }, 
        "y": {
          "inverted": false, 
          "speed": 6000
        }, 
        "z": {
          "inverted": false, 
          "speed": 200
        }
      }, 
      "color": "default", 
      "current": true, 
      "default": true, 
      "extruder": {
        "count": 2, 
        "nozzleDiameter": 0.4, 
        "offsets": [
          [
            0.0, 
            0.0
          ], 
          [
            0.0, 
            0.0
          ]
        ], 
        "sharedNozzle": false
      }, 
      "heatedBed": true, 
      "heatedChamber": false, 
      "id": "_default", 
      "model": "Generic RepRap Printer", 
      "name": "Default", 
      "resource": "http://localhost:5000/api/printerprofiles/_default", 
      "volume": {
        "custom_box": false, 
        "depth": 200.0, 
        "formFactor": "rectangular", 
        "height": 200.0, 
        "origin": "center", 
        "width": 200.0
      }
    }
  }
}
```
By the [docs](https://docs.octoprint.org/en/master/api/printerprofiles.html#sec-api-printerprofiles-datamodel-profile) I was thinking I would get a list that I could iterate over to work out the active profile for example. But I'm not returned a list 🙁 

Present in 1.6.1 & maintenance, not sure if doc bug or unintended behaviour but I have no idea when this changed. I can look at the code to figure out, but putting this issue here first

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

NA

### Version of OctoPrint

1.6.1 & 2.0.0.dev1297+gb133979f2 (I think this is PR #4159 that I checked out)

### Operating system running OctoPrint

OctoPi 0.18 & Windows 10

### Printer model & used firmware incl. version

VIRTUAL

### Browser and version of browser, operating system running browser

Edge 91

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

cp2004, 2021-06-07:
Actually it looks like the data model is correct (https://docs.octoprint.org/en/master/api/printerprofiles.html#profile-list) but it is the wording here that is not great, and the example response is wrong (https://docs.octoprint.org/en/master/api/printerprofiles.html#get--api-printerprofiles):

>Retrieves a list of all configured printer profiles.

> Returns a 200 OK with a list of [profiles](https://docs.octoprint.org/en/master/api/printerprofiles.html#sec-api-printerprofiles-datamodel-profile)

foosel, 2021-06-08:
Sounds indeed like suboptimal documentation phrasing then. Do you have a suggestion for improvement? "collection" instead of list maybe? Or actually "object"?

cp2004, 2021-06-08:
First thing I was thinking of was adjusting the link to the base data model table (this one https://docs.octoprint.org/en/master/api/printerprofiles.html#profile-list), since I initially missed that and skipped straight to the data model of an actual profile. And correct the example response to be inline with current response, so I don't always jump to conclusions...

The wording is difficult... Object representing a list of the available printer profiles? I think the change is very small. Maybe:

> Retrieves an object representing all configured printer profiles.

> Returns a 200 OK with a [profile list](https://docs.octoprint.org/en/master/api/printerprofiles.html#profile-list) response

?

foosel, 2021-06-08:
My working copy is currently pretty preoccupied with half done commlayer stuff, so I did the doc fix in a PR right here on GitHub... Can you do a quick sanity check as well? 😅

commit 4c1152c7 (2021-06-08): 📝 Fix printer profiles API docs

Phrasing improved and some copy-paste issues resolved as well.

Fixes #4160
closed 2021-06-08 by foosel