OctoPrint/OctoPrint #4161: Current layer stats conflict

quadcom, 2021-06-08:
### The problem

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.77 Safari/537.36
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 8204619776
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.4
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.6.1
printer.firmware: Marlin V0.0.2 (Github)
systeminfo.generator: systemapi

At first, I thought this was a problem with the DisplayLayerProgress plugin but on a closer look, this has to be something in OP.

DisplayLayerProgress and the State panel both show the current layer as 16/285 but if you look on the GCode Viewer it states it's printing layer 17. One layer up.

[image]

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

I don't think this is a plugin conflict, currently in print for the next day

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

OctoPi 0.18.0

### Browser and version of browser, operating system running browser

Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) 

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20210608141815.zip](https://github.com/OctoPrint/OctoPrint/files/6616766/octoprint-systeminfo-20210608141815.zip)

Starting dependency resolution...
packed_core.js?7846f86e:1238 ... dependency resolution done
packed_core.js?7846f86e:1261 Initial application setup done, connecting to server...
packed_core.js?7846f86e:1064 Connected to the server
packed_core.js?7846f86e:1069 Triggering connect on all view models
packed_core.js?7846f86e:1071 Server (re)connect processed
packed_core.js?7846f86e:1261 Finalizing application startup
packed_core.js?7846f86e:1242 Going to bind 42 view models...
packed_core.js?7846f86e:1254 Did not bind view model BackupViewModel to target #wizard_plugin_backup since it does not exist
packed_core.js?7846f86e:1254 Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
packed_core.js?7846f86e:1254 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
packed_core.js?7846f86e:1254 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js?7846f86e:220 User pi logged in
packed_core.js?7846f86e:225 Login mechanism: http
packed_core.js?7846f86e:1259 ... binding done
packed_core.js?7846f86e:1259 Application startup complete
6The resource <URL> was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/?#gcode:1 The resource http://192.168.200.244/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/?#gcode:1 The resource http://192.168.200.244/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/?#gcode:1 The resource http://192.168.200.244/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/?#gcode:1 The resource http://192.168.200.244/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/?#gcode:1 The resource http://192.168.200.244/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/?#gcode:1 The resource http://192.168.200.244/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.

cp2004, 2021-06-08:
Please upload the gcode file you have this issue with.

As the note below the gcode viewer states, they have different implementations. Would need to read the actual gcode file to find out which is correct. Note that the gcode viewer takes every line that is at separate Z height as a layer. So if you have a prime line at a different Z height to your 'first' layer then it counts one more.

foosel, 2021-06-08:
Also be advised that your log shows undervoltage issues as recent as yesterday. If not already taken care of, you should fix that ASAP.

quadcom, 2021-06-08:
Here's the GCode

The under-voltage warnings were from old log data running a Pi3b. Currently running on Pi4b with a 3A PSU. There are no warnings under that hardware. I was swapping hardware while troubleshooting a network connectivity issue.

[Desk Drawer.zip](https://github.com/OctoPrint/OctoPrint/files/6616888/Desk.Drawer.zip)

cp2004, 2021-06-09:
So I had a look, the difference just depends on where you start counting from. Display Layer Progress starts at 0, the gcode viewer starts at 1. And by the looks of it in DLP this is determined by the slicer, so it is not in the plugin's control either. There's a setting to adjust this offset in DLP's settings, under the 'Layer' tab:

[image]

*For reference*:
State panel:
[image]

Gcode viewer:
[image]

closed 2021-06-09 by cp2004