OctoPrint/OctoPrint #4162: Gcode files with no extrusion are analysed on every server startup

cp2004, 2021-06-08:
### The problem

I have several gcode files that I use for testing (some of them are [here](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/tree/master/tests/M503)), which have no extrusion but output a bunch of data. Every server startup I see lots of files to be analyzed, but the response is empty. They don't take long to analyse, but it is just unneccessary stuff that gets done on server startup. Not a huge issue in the real world as most people aren't like me with a bunch of empty files sitting around...

### Did the issue persist even in safe mode?

Yes, it did persist

### If you could not test in safe mode, please state why

I run the development env basically stock anyway, with only the plugins I'm working on at the time enabled.

### Version of OctoPrint

Maintenance @ b133979

### Operating system running OctoPrint

Windows 10

### Printer model & used firmware incl. version

VIRTUAL

### Browser and version of browser, operating system running browser

Edge 91

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20210608143536.zip](https://github.com/OctoPrint/OctoPrint/files/6616943/octoprint-systeminfo-20210608143536.zip)

foosel, 2021-06-14:
Side effect of #4061. So... that's going to be icky to solve.

foosel, 2021-06-14:
Less icky than feared.

commit 054b8568 (2021-06-14): 🐛 Set empty analysis result for empty files

That way they won't get added to the backlog again on every single
server startup.

Fixes #4162
closed 2021-10-11 by foosel