OctoPrint/OctoPrint #4165: [Request] RaspPi4 - utilize GPU for webcam processing

quadcom, 2021-06-09:
### Is your feature request related to a problem? Please describe.

Just getting started with printing and found OP which is awesome. 

I set my C920 to 1920x1080/24 and am using Octolapse. I'm hardwired so I have the bandwidth.

Current CPU usage is about 1-4% while printing and capturing a timelapse. But the webcam stream is jittery.

Are there plans to utilize the GPU onboard the Pi4 series as there's a ton of graphical processing support there that I don't think is being taken advantage of? (Correct me if I am wrong there) If a Pi4 can output 4K60 10bit h265 video files without breaking a sweat it would be interesting to see what it can do with a 1080p live stream.

### Describe the solution you'd like

Utilize the GPU for video processing.

### Describe alternatives you've considered

Sorry, I don't have any

cp2004, 2021-06-09:
OctoPrint doesn't actually deal with any webcam streaming at all, it is just embedding a stream into the UI directly into a HTML `<img>` tag. So there's nothing we can do about this *here*.

I assume you are probably using OctoPi, which uses mjpg-streamer. MJPG is not an efficient format like h264/5 is, but mjpg streamer can't output this. Most webcams provide the data in mjpg as well, so you would have to do some transcoding. So CPU-based, mjpg is easily doable on the Pi but it is not efficient for the network.

You can use other streamers - I'm using [ustreamer](https://github.com/pikvm/ustreamer) on mine. The only thing you are limited by is the UI, which supports mjpg or HLS streaming currently.

What you may also find, is that the webcam stream is not stuttering if you look at it outside the UI, at the raw URL of `http://<your-ip>/webcam/?action=stream`. There's a lot going on in OctoPrint's UI and this can cause the browser to slow down and stutter often.

Conclusion: If you find a streamer that does a nice h264 stream, but the only limiting factor is embedding it in the UI (it might need a `<video>` tag instead of `<img>`) then I'll work out support for it. Otherwise there's no streamer in OctoPrint that can be asdjusted.

closed 2021-06-09 by cp2004

quadcom, 2021-06-09:
Hey cp2004

Thanks for getting back to me so quick!

I apologize if I'm off base here. I'm by no means a developer and C+ to me is just a better-tasting orange drink (lol)

I did manage to come across this tidbit about h264 streams coming from cameras (such as the C920, C922) and that there is a library that can grab the data.

If this of any use/help - https://raspberrypi.stackexchange.com/questions/63798/get-h264-from-webcam-logitech-c922

My only goal here is to help as well as I can to make OctPi even better.

cp2004, 2021-06-09:
Better webcam streaming is something that OctoPrint would benefit from, and there must be cameras that work with it - but I'm not sure what that thread is solving, other than 'Logitech were trying to not let people use H264'. I do know that The Spaghetti Detective plugin has a custom built streaming solution ([here](https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/master/octoprint_thespaghettidetective/webcam_stream.py)).

One of the biggest issues is compatibility, MJPG is widely supported, but others are not. If you want to give it a go however, stop by the [discord server](https://discord.octoprint.org) (or the [forums](https://community.octoprint.org)) if you think you have a solution or want help.

reedy, 2021-11-07:
Should we file a task about getting ustreamer support a bit more mainstream in OctoPrint?

https://github.com/guysoft/OctoPi/issues/698 was filed, and ustreamer is now upstream in debian stable - https://tracker.debian.org/pkg/ustreamer

jneilliii, 2021-11-07:
I have a custopizer implementation of this but haven't fully tested.

https://github.com/jneilliii/OctoPi-UpToDateUstreamer/releases