OctoPrint/OctoPrint #4166: [Request] Parse 'Python incompatible' error message from pip and show FAQ link?

cp2004, 2021-06-09:
### Is your feature request related to a problem? Please describe.

When people create plugins that are Python 3 only, we recommend setting the `python_requires` field in `setup.py` to avoid issues and support where people say 'Install was successful but plugin does not load'. It works for many, as they can read error messages and work out what they have to do.

Example response:
```
Installing plugin from https://github.com/jneilliii/OctoPrint-OctoDashCompanion/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmpwlZSIx/master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpwlZSIx/master.zip
ERROR: Package 'OctoDash-Companion' requires a different Python: 2.7.16 not in '>=3.3, <4'
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
```

### Describe the solution you'd like

If OctoPrint could show a notification or something at that point with a link to an FAQ post (I can write it with relevant links to my blog post & upgrade script) this may help users know what to do and get more people upgraded.

Some kind or parsing on `requires a different Python:`? This might work.

### Describe alternatives you've considered

It might be a stretch, but I think we should start warning users on Python 2 that it is EOL, no longer supported and they should upgrade ASAP via a (dismissible) warning in the UI. We got to 46% but progress has slowed. It would enable the update to 2.0.0 to run smoother. Maybe some people might be annoyed and not want to update, but then the alert is dismissible and so they should be OK just clicking an `X`.

foosel, 2021-06-14:
### Python version mismatch handling

[image]

[image]

Will also be displayed when trying to install a single file plugin that has an incompatible `__plugin_pythoncompat__`.

### Py2 update nag screen

[image]

I'd like to take you up on your offer to write an FAQ entry now 😉

cp2004, 2021-06-14:
> I'd like to take you up on your offer to write an FAQ entry now 😉

On the way to a forum near you some time this weekend 🙂

foosel, 2021-06-22:
@cp2004 poke? :D

cp2004, 2021-06-22:
Most of the way written at the weekend locally, will finish it now...

foosel, 2021-06-22:
Dude, if you need another weekend, no harm done, I just wanted to make sure it doesn't get forgotten (also because I still need to set up an FAQ redirect and stuff)

cp2004, 2021-06-22:
Live here https://community.octoprint.org/t/upgrading-your-octoprint-install-to-python-3/35158

foosel, 2021-06-22:
And now redirected to from https://faq.octoprint.org/python3-update

foosel, 2021-06-22:
And https://faq.octoprint.org/plugin-python-mismatch now also leads somewhere.