OctoPrint/OctoPrint #4167: Server Folder Path's reset to defaults on system boot

SaKiEQ, 2021-06-10:
### The problem

Customized folder paths for upload, time-lapse etc. reset to default (on pi storage paths) after reboot of the system. 

I am 'trying' to have time-lapse stored on a NAS via network mount. All works well until the pi / octoprint gets restarted and everything returns to defaults... 

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

-

### Version of OctoPrint

Version 1.6.1

### Operating system running OctoPrint

browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_13_6) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/13.1.2 Safari/605.1.15 connectivity.connection_check: 1.1.1.1:53 connectivity.connection_ok: true connectivity.enabled: false connectivity.online: true connectivity.resolution_check: octoprint.org connectivity.resolution_ok: true env.hardware.cores: 4 env.hardware.freq: 1500 env.hardware.ram: 1903697920 env.os.bits: 32 env.os.id: linux env.os.platform: linux2 env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2 env.plugins.pi_support.octopi_version: 0.17.0 env.plugins.pi_support.throttle_state: 0x0 env.python.pip: 19.3.1 env.python.version: 2.7.16 env.python.virtualenv: true octoprint.safe_mode: false octoprint.version: 1.6.1 systeminfo.generator: systemapi

### Browser and version of browser, operating system running browser

OS X, Safari

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

...

SaKiEQ, 2021-06-10:
I checked 'systemctl' and the mount loads before octoprint service engages. 

The same happens when i only restart octoprint, without reboot of the system. For some reasons the customized paths are not stored and defaults are load on each start. 

Of corse I can dig-in and modify the defaults with my custom paths, but I think that would the wrong way to address the issue :)

SaKiEQ, 2021-06-10:
[image]

all rights to the mount seem to be fine, checked via terminal and tested when Octoprint is storing a time-lapse on the NAS. I can rule out that it is a revert-to-defaults because of faulty config.

foosel, 2021-06-10:
Sadly you forgot to upload the requested systeminfo bundle, so it's impossible right now to check the logs and confirm that yes, this default reverting is indeed because OctoPrint can't confirm that it can actually write to your NAS mounts on startup.

SaKiEQ, 2021-06-10:
added the systeminfo bundle in original post, please check. 

What would I have to change that Octoprint can check on startup that writing is possible ? Else custom path's are relatively useless if they can only be set at runtime... specially since I am technically using more then just the time-lapse path, the uploads and watch path too.

foosel, 2021-06-10:
Can you reproduce the error, then share another systeminfo bundle? From the looks of it, the issue hasn't happened yet today during startup and the only errors that look file system related are these:

```
2021-06-10 16:35:44,567 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from https://plugins.octoprint.org/plugins.json
2021-06-10 16:40:14,404 - octoprint.server.api - ERROR - Error while trying to create /home/pi/NASstorage/Print\ Files/Ender3
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 484, in _test_path
    os.makedirs(path)
  File "/home/pi/oprint/lib/python2.7/os.py", line 150, in makedirs
    makedirs(head, mode)
  File "/home/pi/oprint/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 22] Invalid argument: '/home/pi/NASstorage/Print\\ Files'
2021-06-10 16:40:34,831 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-06-10 16:41:28,031 - octoprint.server.api - ERROR - Error while trying to create /home/pi/NASstorage/Video\ Files/Ender3
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 484, in _test_path
    os.makedirs(path)
  File "/home/pi/oprint/lib/python2.7/os.py", line 150, in makedirs
    makedirs(head, mode)
  File "/home/pi/oprint/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 22] Invalid argument: '/home/pi/NASstorage/Video\\ Files'
```

However, they aren't from the actual server startup. Could be related however.

Custom paths definitely work if the system is properly configured and the mounts are available, properly permissioned and actually allow writing on startup. OctoPrint tests all of this since in the past we had people complaining about their servers no longer starting up after they added network mounts that were misconfigured: 

https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/settings.py#L2222-L2272

SaKiEQ, 2021-06-10:
have you noticed the 'double back-slashes' in the path ? I noticed that when enter the path, trailing empty spaces '\ ' the entry field rejects my path, requiring me to enter the 'human readable' version of the path, so somehow the path in the log receives a second '/home/pi/NASstorage/Print\\ Files' after /Print, at least in the error log.

SaKiEQ, 2021-06-10:
Pre-Reboot:
-changed path for 'uploads' with full path
-changed path for 'time-lapse' with relative user path

[octoprint-systeminfo-20210610172650.zip](https://github.com/OctoPrint/OctoPrint/files/6632155/octoprint-systeminfo-20210610172650.zip)
[image]

SaKiEQ, 2021-06-10:
@foosel  I think the error that you have spotted, was when I was trying to enter the path in the different versions, while using the 'test path' button. I have only restarted on tested path configurations that returned 'path is valid'. Most likely prior to that I tried to enter the path with single backslashes which threw the error in the log as you highlighted. 

In the above two logs - prior and post restart I can not spot any errors or warnings, yet the path's are reset to defaults

foosel, 2021-06-10:
Yeah, I can't see any either, which confuses me. Looks like for some reason logging of the folder validation is disabled here, which of course isn't exactly helping.

Are you comfortable with running a version directly from git? If so I could fix the logging situation and then we might get enough info to figure out what exactly is bothering it here.

SaKiEQ, 2021-06-10:
Ok - when i change the file paths, to such that are not using any empty spaces as shown in screen shot, then the entered custom path's survive restart / reboot ! 

Leading me to believe, the there is a bug in the file path validation, since it won't accept me to enter manually a path in the format of '/home/pi/NASstorage/Print\ Files/Ender3' - it only accepts '/home/pi/NASstorage/Print Files/Ender3', but then discards the path on restart...
[image]

SaKiEQ, 2021-06-10:
Addition: I of corse created the according  folder without spaces on the NAS as-well. 

The "Print Files" version throws the  error, the "Print_Files" works after restart. 
I have tried to enter file path as with backslash, in quotation marks, but neither is validated. Only the plain text version with empty spaces is validated by the test, yet that is the only version that does not survive reboot. 

Using a path that does not contain empty spaces solves the problem. 

[image]

foosel, 2021-06-10:
I cannot confirm a general issue with spaces in the folder names. I just reconfigured a stock 1.6.1 install here to use `/home/pi/.octoprint/uploads 2` instead of `/home/pi/.octoprint/uploads` and that survived a reboot just fine.

SaKiEQ, 2021-06-10:
Interesting ! I tried now several versions and only the one without spaces seems to go through. In command line I can navigate which ever version, mount and unmount - as long as format it the right way. 

This mount is done via CIFS and resides on SMB volume. So maybe the different favors of file formats work  against one another !? 

My 'underscored', non-spaces version, survives restart of the service and reboot of the entire system so far. 

Do you wanna leave it as it is for now ? I am fine by just changing my folder naming, yet I have the tendency to 'need to know' :)

foosel, 2021-06-10:
It might indeed be a CIFS/SMB issue at play here... My current idea it to leave it as is for now, but patch the logging behaviour for 1.7.0 so we'll get some information as to WHY stuff gets rolled back on start, and then if there's indeed a bug we'll hopefully be able to figure it out then. 

This patching idea is also why I asked if you'd be comfortable running a dev branch from git (which would boil down to a handful of commands to activate and deactivate again), because then we could shorten the turnaround here and figure out the precise error that's rearing its head here earlier and maybe already solve it for 1.7.0.

SaKiEQ, 2021-06-10:
Yep - just to confirm, I have done the same on my setup. 

It survives the reboot too. 

What throws a fit, is when it is a remote volume. So I will try and mount and HDD directly on a USB port, go through the same... to isolate it whether is an issue with this NAS, CIFS or else. 
[image]

SaKiEQ, 2021-06-10:
Sure I can do some git-magic, as long as you tell me what to do :-> 

In general I am comfortable on Linux & Co... just don't make me use vi over nano !!!!

foosel, 2021-06-10:
I've just pushed a change to `maintenance`/future 1.7.0 that should enable and improve logging (I also figured out why it wasn't logging like it should have, another sanity check for a duplicated folder paths ran before the actual one, had the same validation and fallback side effects but didn't have logging enabled). 

You can checkout and install `maintenance` on your Pi with these commands:

```
~/scripts/add-octoprint-checkout
cd ~/OctoPrint
git checkout maintenance
~/oprint/bin/pip install -e .
sudo service octoprint restart
```

Once the server's restarted, verify that it says something along the lines of `OctoPrint 1.7.0.dev146+gdf5f45607` in the bottom left corner of the UI, then run a test to reproduce, create a systeminfo bundle and share it here please.

When you want to revert to the stable release:

```
~/oprint/bin/pip install OctoPrint==1.6.1
sudo service octoprint restart
```

SaKiEQ, 2021-06-10:
Cool. 

In the mean time I ran the test with a USB flash drive mounted on the same unit, mkdir /USB-Drive/Print Files... 

That folder on the flash drive survived the reboot, fstab'd the USB drive to mount at boot and still survived after full reboot. 

So the problem is the SBM/NAS version of the file path... Maybe something in the host name resolution between python and NAS

SaKiEQ, 2021-06-10:
I do think I have circled the bug, it is fortunately not with Octoprint !!

The QNAP Nas seems to have an issue in their SMB implementation when it comes to file path with empty spaces, regardless how I format the path, it stops and breaks at the first empty space in the string. Hence returning an incomplete path, or folder not found !

SaKiEQ, 2021-06-10:
yep, '/home/pi/NASstorage/Print%20Files/Ender3' does work and survive, despite QNAP being a linux flavor, the file system is some sodomized SMB / Windows implementation that does not follow in every case standard formatting. 

When used manually via terminal commands backslashes work fine and I can navigate to the remote folder, create dummy files from the Pi on the NAS etc.. The moment a system process (or python, or rsync) does the same it breaks with \ or with ' ' ! %20 does seem to work in either case. 

Also you sanity check of the path in Octoprint is fine with %20 as space marker.

SaKiEQ, 2021-06-10:
ohhhh... one more thing.... It worked from a Octoprint perspective... yet NAS get folders like this 
[image]

SaKiEQ, 2021-06-13:
Hello, 

I ran your 'fix' from Git despite having fixed it with a underscore instead of space in 'Print_Files' path.

Attached the system info post that. Please note that tI have messed around with the file path, by adding backslashes instead of empty spaces or underscores > Hence the error at the end of the log. It does not like backslashes now, throws an path error.

In addition it seems to validate the path properly now, also it creates an empty folder in the destination if it does not exist.

[octoprint-systeminfo-20210613120857.zip](https://github.com/OctoPrint/OctoPrint/files/6643608/octoprint-systeminfo-20210613120857.zip)

BeatSlayer, 2021-06-13:
It seems like a Unicode issue.

foosel, 2021-06-14:
@SaKiEQ May I ask why you reopened this after confirming it's behaving as expected now, validating the path properly?

The error when you add backslashes is correct - the backslahes are escaping necessary on certain shells, the paths themselves don't contain backslashes and adding double escaped backslashes causes path names that are apparently not supported by the OS here:

```
OSError: [Errno 22] Invalid argument: '/home/pi/NASstorage/Video\\ Files'
```

This is not OctoPrint's fault, but it handles it correctly.

Creating the folder is correct and expected behaviour as well.

So, to reiterate, why the reopen?

foosel, 2021-06-14:
@BeatSlayer No, it doesn't.

SaKiEQ, 2021-06-14:
Hi there, I felt I owed you the answer of the more detailed logs and until you could not confirm that you you are satisfied with the conclusion - as you just gave above - the issue might better stay open until all boxes are ticked ! 

The OS thing makes sense, one part I tend to forget since in my limited world OctoPrint runs on a pi, with one version version of Pi - which of corse is wrong. 

One thing that puzzles me slightly is, that is creates a folder when no folder is at the destination of the path. Which of corse safes me trouble of errors and increases stability. In my case of trial and errors, it creates every iterations of folders at the destination that I did not necessarily want there... But then again, the QNAP NAS seems to have a lot of peculiarities in its SMB implementation that I was not previously aware of. 

It works now - we got our answers - so time to close again ? 

Thanks anyway :)

closed 2021-06-14 by SaKiEQ