OctoPrint/OctoPrint #4168: Setting 'Confirm before cancelling a print' does not enable confirmation

bluenenschloss, 2021-06-12:
### The problem

Enabling 'confirmation before canceling a print' cancels printing w/o asking for confirmation before

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

linux Ubuntu 18.04 LTS

### Printer model & used firmware incl. version

Prusa-Firmware 3.10.0 based on Marlin

### Browser and version of browser, operating system running browser

Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.1.1 Safari/605.1.15

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

jneilliii, 2021-06-12:
please provide system info bundle as requested, don't just check the box.

bluenenschloss, 2021-06-12:
Getting unreliable results after retesting - closing this issue until further investigation finished.

closed 2021-06-12 by bluenenschloss