OctoPrint/OctoPrint #4169: Cant revert back to after temp maintenance install

SaKiEQ, 2021-06-13:
following your instructions, I tried to revert back to 1.6.1 after the 1.7.0 maintenance test. OctoPrint fails to launch ! 

I went back to the Git maintenance update and it starts once again... 

~/oprint/bin/pip install OctoPrint==1.6.1

SaKiEQ, 2021-06-13:
System info post 1.7.0 maintenance update, failed reverting back to 1.6.1 and back to 1.7.0 maintenance. 

[octoprint-systeminfo-20210613122759.zip](https://github.com/OctoPrint/OctoPrint/files/6643624/octoprint-systeminfo-20210613122759.zip)

foosel, 2021-06-14:
Your log indicates two files missing after the revert that should be there (and are when I do the same upgrade/downgrade cycle on my side). Please make sure that your filesystem isn't full, and try again, forcing a reinstall:

```
~/oprint/bin/pip install --force-reinstall OctoPrint==1.6.1
```

SaKiEQ, 2021-06-14:
Thanks, could you tell me which files went missing ? Just so it gives me a clue where I might have messed up something unintentionally ! 

I could identify it from logs my self of corse, but since you might remember which ones that might safe some hide and seek time.

foosel, 2021-06-14:
It jumps right at you in your log on bundleviewer:

[image]

But that's definitely not something you should have accidentally deleted, some kind of file corruption or something similar is more likely, so please try the steps I outlined above.

SaKiEQ, 2021-06-14:
Cool - thanks - yeah looks better in budleviewer.... I am using BBedit -
old school.

The machine ran out of card memory last year, was since then recovered and
upgraded to larger card.
In this period now of tinkering around there was more then enough left on
this card for not being the culprit .

I will try the above steps in a bit - printing right now :)

SaKiEQ, 2021-06-14:
Wild - the installation process works fine - yet on restart the local OctoDash client can connect to octoprint, while I only get the "Connecting to OctoPrint Server" idle wheel page... forced reload, trashing the browser cash did not resolve it. 

When I revert back to the maintenance install, everything goes back to normal - Yet OP tells me there is a never version '1.6.1' available to install. 

Tried the update install  via OP, with same result !

Right now I am 'stuck' on the maintenance install. Funny enough, after the full re-install I get empty octoprint log, as if the service did not load. Yet systemctl tells me the process is running. The served 'Connecting to' webpage is a sign that the service is running... weird  ?

foosel, 2021-06-14:
That... really doesn't make sense at all. Something appears to be seriously wrong with your setup (and could have broken at a regular update as well in all likelihood). For reference, I habitually switch back and forth between branches, stable releases and such without *any* issues at all, it's part of my regular development workflow.

Can you please run `~/oprint/bin/pip install --force-reinstall OctoPrint==1.6.1` followed by a `sudo service octoprint restart` again and share the full output?

SaKiEQ, 2021-06-14:
I think I am about to give-up, erase the pi and start over... This is my oldest Pi and most likely carries a legacy of screw-ups from my side.

I tried now several iterations of installing the previous version, udating/grading the entire system, pip and all... In 1.6.1 I get the same problems over and over again. 

I get the login/authentication web page, followed by "Connecting to Server..." and that is where it stops. 
on the 1.7.0 version I get all the way back to the normal UI. 

Most likely there is an issue with ownership/permissions somewhere along the trail of pip, which prevents it from re-installing everything. But why does 1.7.0 load all the way, when 1.6.1 can't ? 

I guess its time to write down all the configs of all plugins and begin to erase the Pi... ?
[Uploading octoprint-systeminfo-20210614142540.zip…]()

foosel, 2021-06-14:
You can of course always start over, but I'd seriously prefer if you'd do:

> Can you please run `~/oprint/bin/pip install --force-reinstall OctoPrint==1.6.1` followed by a `sudo service octoprint restart` again and share the full output?

Something somewhere goes wrong, because the above should reinstall *everything* without any stuff missing or corrupted or weird, and it doesn't. And I can't tell you why 1.7.0.dev works and 1.6.1 doesn't when I don't know what the output of the above is.

SaKiEQ, 2021-06-14:
Well when I run ~/oprint/bin/pip install --force-reinstall OctoPrint==1.6.1, then I can can't get to the UI... hence I can't safe the bundle out. As mentioned above the log in ~/.octoprint/logs/octoprint.log is empty after the update and restart of service and/or server. 

In the above attached log bundle, it shows the system start under 1.6.1 and then followed by 1.7.0 after the next restart, so I hoped that would give you some clues. 

There are some discrepancies in the log, from 1.6.1 (which i updated with your above command) and the 1.7.0. 
What you see in the log is literally a full install of 1.6.1 with your above command. Then when I was unable to log-in/load the UI, then I would revert to the 1.7.0 version.

FYI#1  I am manually shutting down the octoprint service 'sudo service octoprint stop', perform the update, reboot the pi. 
I tried the 'sudo service octoprint restart' without reboot, with same result. 

FYI#2 I am getting a PIP message, to upgrade to the latest version, which I did, but even after reboot it keeps coming back. 

My conclusions for the moment there is something happening between python & pip....

foosel, 2021-06-14:
I'm not talking about a bundle though. I said I want the **output** ;) Like, what it writes on the screen. Additionally, you can get a bundle even if the UI doesn't work: https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887

SaKiEQ, 2021-06-14:
achso.... :)

SaKiEQ, 2021-06-14:
[Uploading 1.6.1 install teminal dump.txt…]()

foosel, 2021-06-14:
That upload didn't actually make it through.

SaKiEQ, 2021-06-14:
ok we try again... today its like a bee-hive here, so my focus is crap... and the results too. 

[1.6.1 install teminal dump.txt.zip](https://github.com/OctoPrint/OctoPrint/files/6648502/1.6.1.install.teminal.dump.txt.zip)

foosel, 2021-06-14:
Well, there we have it:

```
ERROR: Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/home/pi/oprint/lib/python2.7/site-packages/~ytz-2021.1.dist-info/RECORD'
```

Permission error, probably due to you running sudo at some point where it was not only not necessary but harmful.

Try this: 

```
sudo chown -R pi.pi /home/pi/oprint
```

Then try the reinstall again. If there are any errors, *share them*.

SaKiEQ, 2021-06-14:
yeah well... did that, no error during install, but...

Again it won't allow me past user login / Connecting to Octoprint Server. 

[1.6.1 install teminal dump 2.txt.zip](https://github.com/OctoPrint/OctoPrint/files/6648724/1.6.1.install.teminal.dump.2.txt.zip)

Please find the system info bundle once again with current state of affairs. 
[octoprint-systeminfo-20210614165658.zip](https://github.com/OctoPrint/OctoPrint/files/6648842/octoprint-systeminfo-20210614165658.zip)

SaKiEQ, 2021-06-14:
I am slowly tracing the errors on my system. 

IOError: [Errno 2] No such file or directory: '/home/pi/OctoPrint/src/octoprint/static/js/app/bindings/copywidth.js'

Does not exist, I maneuvered manually to the folder and looked for the file. 

I am going to install the 1.7.0 once again over it, check the same file path and see if there is something changing.

SaKiEQ, 2021-06-14:
After '~/oprint/bin/pip install -e .' the system connects as on all previous attempts, the file in the above IOError path still does not exits, but the server starts and connects.

I assume there is file.library dependency that is different in the 1.7.0 release compared to the 1.6.0. 
Maybe I would have to roll back way later then 1.6.1 to get the missing files installed, they might be just legacy and still linger around, but not actively maintained in the install / update ? 
But then what would happen on a clean install with the same installer script ?

foosel, 2021-06-14:
On closer look, that file (and the other one) still points to the 1.7.0.dev install. It shouldn't be trying to even fetch that under 1.6.1, let alone with a path completely outside of the install location `~/oprint/...`. It's also no wonder that it doesn't exist on 1.7.0.dev given that these files were removed in 1.7 due to no longer being used. 

Final idea, then I'll have to recommend a reinstall because I'm frankly out of ideas of WHAT is broken there on your install:

  1. `sudo service octoprint stop`
  2. Make sure the server is actually stopped: `ps -ef | grep -i octoprint` should only return the `grep` command itself
  2. Run `~/oprint/bin/pip uninstall OctoPrint` until it tells you it is no longer installed (depending on what happened to this installation in the past, that might have to be repeated a few times)
  3. Delete the generated web assets: `rm -rf ~/.octoprint/generated`
  4. Reinstall 1.6.1 once again: `~/oprint/bin/pip install --force-reinstall OctoPrint==1.6.1`
  5. `sudo service octoprint restart`
  6. Clean the cache in your browser
  7. Try to load

Again, I cannot reproduce that, I've been happily rolling back and forth between 1.6.1 and 1.7.0.dev and a number of other versions without any issues.

SaKiEQ, 2021-06-14:
>    3. Run ~/oprint/bin/pip uninstall OctoPrint until it tells you it is
>    4. Delete the generated web assets: rm -rf ~/.octoprint/generated
>    5. Reinstall 1.6.1 once again: ~/oprint/bin/pip install
>    6. sudo service octoprint restart
>    7. Clean the cache in your browser
>    8. Try to load

Ok, what will happen with all the plugins in python2.7/site-packages when I
do that  ?

foosel, 2021-06-14:
Nothing.

foosel, 2021-06-14:
Side note, I'm closing this so no one gets confused in the bug tracking here, as it is clearly not a general bug but rather an issue with your environment and thus a support request, not a bug report. Normally I'd redirect this to a forum post, but given that we already started out here we can continue here.

This being closed doesn't mean comments are closed too, it is merely a state flag to signify this is not actionable from OctoPrint's side.

closed 2021-06-14 by foosel

cp2004, 2021-06-14:
I have seen this happen around the forums/discord server once or twice, whenever we add whole static files, pip does not seem to fetch all the necessary static files (no idea why) but this command always fixed it:
```
~/oprint/bin/pip install --force-reinstall --no-cache-dir OctoPrint
```
Never had any further problems from anyone reporting similar issues (missing static files) after suggesting that. Definitely outside OctoPrint's control, possibly permissions or bad download - I would think pip should have some kind of validation/error handling there, but I have seen some *weird* issues in the past, such as one where pip would replace the first character of the package with a `~` and so it would not be properly uninstalled.

You can take a look, by running something like `ls ~/oprint/lib/python3.7/site-packages/ | grep -i print` and seeing if any of the names are corrupted. Not likely, but just evidence that sometimes pip breaks when switching between install locations (git vs. pypi).

SaKiEQ, 2021-06-14:
Ok, but Python3.7 is not installed on this pi, I am on Python2.7 as far  I
remember. ~/oprint/lib contains only python2.7.

Anyway I think its a waste of time by now, it will be a few hours to
restore all the plugins and connectivity on a fresh install, but right now
 I am not sure if
this will yield a safe restoration of 1.6.1.

I tried all the above in several combinations. Purged Pip caches, removed
and reinstalled python2.7 just to make sure that permissions and rights are
restored.

Un the un-install there as an error of a file that could not be deleted, so
I 'sudo rm'd' it out of the way and ran the de-installer again.

Tried now the new install and it throws the same errors as before.

It must be some user rights that are screwed up somewhere that plays a
curveball on pip.

SaKiEQ, 2021-06-14:
Do you have any good idea, why I would have a /oprint on root level, not /home/pi/oprint as expected (i have one there too). 

It seems to have all the site-packages in it that i would expect. This is the first time I notice this, somehow, some path got really messed up. 

So I think this really needs to be re-installed from scratch to be safe. 

But thanks a lot anyway :)

jneilliii, 2021-06-14:
> Do you have any good idea, why I would have a /oprint on root level, not /home/pi/oprint as expected (i have one there too).
>
> It seems to have all the site-packages in it that i would expect. This [...]

usually happens when it's installed outside of the virtualenv folder (oprint) or it's not activated (source ~/oprint/bin/activate) or possibly from using sudo command.

foosel, 2021-06-14:
> Do you have any good idea, why I would have a /oprint on root level, not /home/pi/oprint as expected (i have one there too).

At some point, you used `sudo` or worse, ran an install or something as `root` right away for anything else but starting/stopping the server.

SaKiEQ, 2021-06-14:
right -hence the root path. Must have been a long time ago... funny how this thing held together so long and only broke now.

SaKiEQ, 2021-06-14:
Thanks for all your help and effort ! I think I learned more about OctoPrint and its inner workings in the past 12h then in the past 12month :)

I reinstalled everything apart from the main Raspi OS... so far so good. I am back on 1.6.1 ! 

I pulled a backup of my config before, but apparently I was on 1.7.0 dev version... OP complained about the version, so I tried to modify the .json's with appropriate version numbers. No luck - I think you are checksumming/hashing the config ? 

I am getting a error 'no metadata.json' despite it being present - I guess thats a blanket error for bad format / 
[Creality-Ender-3-V2.zip](https://github.com/OctoPrint/OctoPrint/files/6650765/Creality-Ender-3-V2.zip)