OctoPrint/OctoPrint #4170: [Request]Disable Stepper Driver Motor

dumbozo, 2021-06-14:
### Is your feature request related to a problem? Please describe.

On the Ender 3v2 (but probably most printers) you can disable the stepper driver motors so you can manually change x, y and z-axis. This feature is not found in Octoprint so I have to turn to the printer display to disable these while other functions as extrusion, homing, changing x, y an z-axis through GUI is possible

### Describe the solution you'd like

just a button in the control page which disables the stpperdriver motors.

### Describe alternatives you've considered

use printer display

closed 2021-06-14 by dumbozo

cp2004, 2021-06-14:
This already exists:
[image]

dumbozo, 2021-06-14:
just see motors off... hmm wonder if I am so blind... lol

dumbozo, 2021-06-14:
sorry and thank you... I've clearly missed it while now looking for a logical location I found it. I think the bells din't ring... just looking for stepper driver or so