OctoPrint/OctoPrint #4171: custom controls - regex not matching

dwery, 2021-06-15:
### The problem

I have a custom control configured like this:
```
 - name:MyLabel
    regex: Probe Offset.+Z([0-9\.-]+)
    template: 'Z Offset: {0}mm'

```

The control is shown, but the content is never updated. It is supposed to parse the output of M851
`Probe Offset X-44.10 Y-10.80 Z-0.92
`

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

Linux 5.10.39

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20210614222603.zip](https://github.com/OctoPrint/OctoPrint/files/6653025/octoprint-systeminfo-20210614222603.zip)

cp2004, 2021-06-15:
Seems to work fine for me, copied the above configuration snippet into config.yaml, sent `M851` and the offset shows up. Also simulated the line you pasted above with the virtual printer and it seems to work just fine.

Could you enable `serial.log`, try sending M851 and then upload the resulting log here? So we can check the regex is right for what is coming out of the printer?

dwery, 2021-06-15:
@cp2004 here you go:
`2021-06-15 12:14:48,085 - Recv: Probe Offset X-44.10 Y-10.80 Z-0.92
`

cp2004, 2021-06-15:
[image]

Definitely works for me then. Not sure what to do about this one, other than leave it and see if anyone else has the issue, or if you find some kind of misconfiguration or conflict with another control.

dwery, 2021-06-15:
I tried in safe mode, is there anything that safe mod does not disable?

foosel, 2021-06-22:
> is there anything that safe mod does not disable?

[Safe mode disables third party plugins](https://docs.octoprint.org/en/master/features/safemode.html). It does not revert any kind of settings though.

foosel, 2021-06-22:
For the record, I cannot reproduce it either. Line gets parsed and output updated just fine.

dwery, 2021-06-22:
I must have done something wrong with safe mode. I tracked down the issue to the calibration companion plug-in.

closed 2021-06-22 by dwery

foosel, 2021-06-22:
Glad you found the issue! 😊

cp2004, 2021-06-22:
> I tracked down the issue to the calibration companion plug-in.

That should now be fixed with the latest version of that plugin, see https://github.com/iFrostizz/OctoPrint-CalibrationCompanion/pull/19