OctoPrint/OctoPrint #4174: [Request] Add 'Access-Control-Allow-Credentials' header when CORS is enabled

cp2004, 2021-06-19:
### Is your feature request related to a problem? Please describe.

When trying to create web apps to use OctoPrint's API, I can't use any session cookies from the login, as the above header is not sent by OctoPrint. When trying to do something like this:
```js
fetch("http://localhost:5000/api/login", {credentials: 'include', method: 'POST', body: JSON.stringify({user: "Charlie", pass: "Pudding26"}), mode: 'cors'})
```
Below error occurs:
```
Access to fetch at 'http://localhost:5000/api/login' from origin 'http://localhost:3000' has been blocked by CORS policy: The value of the 'Access-Control-Allow-Credentials' header in the response is '' which must be 'true' when the request's credentials mode is 'include'.
```

So, you must use the API key instead. 

### Describe the solution you'd like

Set `Access-Control-Allow-Credentials` to true when CORS is enabled, allowing for web apps to authenticate using username/password combinations.

### Describe alternatives you've considered

Asking the user for the API key instead of their username & password. This does still work.

### Additional context

Question: Is this how this is supposed to work? If the username/password is *only* intended to be used for OctoPrint's UI and API keys for everything else, then this should probably be documented somewhere.

https://community.octoprint.org/t/api-auth-question/35088/6

foosel, 2021-06-21:
> Question: Is this how this is supposed to work? If the username/password is _only_ intended to be used for OctoPrint's UI and API keys for everything else, then this should probably be documented somewhere.

No, this is not intended. I simply never tested a client via CORS with anything but an API key.

foosel, 2021-06-22:
And after giving this some thought and implementation I'm now wondering if that shouldn't rather be a setting, as it opens up the possibility of CSRF...

cp2004, 2021-06-22:
I'm no web-security expert, but I think I see it- and if it is considered insecure, you don't want to do it, then there is no point doing it, and it means that we always need an API key instead, for all cross origin (browser based) requests.

foosel, 2021-06-22:
I mean... it would allow this from an arbitrary website: 

``` json
fetch("http://octopi.local/api/connection", {credentials: 'include', method: 'POST', headers: { "Content-Type": "application/json" }, body: JSON.stringify({"command": "disconnect"}), mode: 'cors'})
```

And if you'd have CORS enabled and your browser had a session for `octopi.local`, you'd now have just disconnected your printer. Or worse. So yeah... I guess this is rather a documentation bug after all and we'll require API keys for all matters CORS.

cp2004, 2021-06-22:
OK I have clarified how I thought it worked. And yeah, this is not secure. Somewhere I didn't connect it picking up the session from the actual UI, and that it would have to be one created on that site, but no, this is not how these things work. I don't think it's worth enabling this, lets just document that you need an API key.

closed 2021-06-22 by foosel

cp2004, 2021-06-22:
Seems good to me 👍

foosel, 2021-06-22:
It's now also live (I pushed this to `master`, sync to `maintenance` will happen automatically)

commit 85f6fa01 (2021-06-22): 📝 Clarify cookies vs CORS

Closes #4174