OctoPrint/OctoPrint #4178: Invalid temperatur axis labels after printer disconnect during print

zMartini, 2021-06-24:
### The problem

After the printer accidently disconnected during print by manual overides at the printer, the browser display shows invalid temperature axis labels on the OctoPrint Temperatur tab (180000000 degrees). These invalid peek values do NOT reset after printer reconnect and starting a new print. After the peek scrolled out of the window, normal temperature values were displayed. See attached screenshot for an example.

I was not able to reproduce this issue manually to generate a serial communication or javascript log file.

[image]

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

I was not able to reproduce this issue manually.

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

OctoPi 0.17.0

### Printer model & used firmware incl. version

Anycubic Mega X & Firmware Knutwurst v1.1.7 (Marlin 2.0.5.4)

### Browser and version of browser, operating system running browser

Windows 8.1 & Chrome v91.0.4472.114 64-bit

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20210624112844.zip](https://github.com/OctoPrint/OctoPrint/files/6708441/octoprint-systeminfo-20210624112844.zip)

cp2004, 2021-06-24:
There's nothing we can do about this issue here without the serial log really. I can only assume that the printer reported some strange temperatures - my best theory is that there was a bug in the firmware (or some hardware issue), and the printer noticed the temperature was shut down & disconnected due to the error.

I'm going to close this, because there's nothing we can do without the printer communication at the time. The temperature graph should just report what it has received, and as far as I know it is working as intended. You can enable serial.log, and then if it happens again leave a comment with the new log files and we can take a look.

closed 2021-06-24 by cp2004

zMartini, 2021-06-25:
Thank you for your quick reply. While I understand your point about solving this problem, I disagree that there is nothing that can be done. A temperature of 180000000 degrees is pretty unlikely, so some sort of clipping temperature values (example: temp = max(temp,maxtemp), where maxtemp = 1000) would at least keep the scale intact.