OctoPrint/OctoPrint #4179: [Request] Gcode Tab Highlight Center of Bed Render Option

alsmithson, 2021-06-28:
### Is your feature request related to a problem? Please describe.

In the gcode tab, it is hard to determine where an object is located on the bed because all the grid lines are the same.

### Describe the solution you'd like

I sometimes use the gcode tab to view the placement of an object, which is usually near the center of the bed. I currently have to count squares from the front and left side. But if the center gridlines were highlighted with slightly darker or wider lines, I could more easily determine the distance of the object from the center of the bed. You could also highlight every 25, 50, or 100mm line for an even better user experience.

### Describe alternatives you've considered

None

### Additional context

Accurately determining the location is important when manually doing sequential printing. I often do that when fine-tuning a new filament. I could do sequential printing via slicer settings, but that takes a while to setup. Sometimes I just want to do 2 or 3 prints and don't want to go through all the setup in the slicer.

foosel, 2021-10-07:
I'd rather like to go with visually distinguishing 50mm segments here - center lines don't necessarily need to line up with the grid lines here since a bed size isn't required to be a multiple of 10mm, and that would most likely look confusing. 

---

Grid steps in rectangular and circular grid are currently 1cm. Adjust rendering routines for both to render a slightly wider line every 5 steps = 5cm (`1.5 * lineWidthFactor` maybe?).

Related:

  * https://github.com/OctoPrint/OctoPrint/blob/5f7679b11baafc0bb58861c462b9e4568041080b/src/octoprint/plugins/gcodeviewer/static/js/viewer/renderer.js#L505-L630

Minor improvement -> `maintenance` branch

foosel, 2021-11-04:
#4287 implementing this has been merged and will be part of 1.8.0

cp2004, 2022-05-17:
1.8.0 has been released 🎉

closed 2022-05-17 by cp2004