OctoPrint/OctoPrint #4183: [Docker] Older version scheme breaks auto-updaters

Ornias1993, 2021-07-01:
Currently the dockerhub containers an old version scheme following the following format:
5512.0.0

Basically:
Highnumer.0.0

This leads to a situation where auto-updating systems like Renovate, will assume those old version numbers are newer than the actual latest release. (as 5512.0.0 is a higher version number than 1.6.1).

I would request those versions to be removed or retagged in such a way that they won't cause issues with updating systems.

foosel, 2021-07-01:
Wrong repository, the one related to the docker builds is [here](https://github.com/OctoPrint/octoprint-docker).

closed 2021-07-01 by foosel

Ornias1993, 2021-07-01:
> Wrong repository, the one related to the docker builds is [here](https://github.com/OctoPrint/octoprint-docker).

Thanks, sorry for the misunderstanding (octoprint/octoprint and all)