OctoPrint/OctoPrint #4188: [Request] Add operative temperature range option

ElDavoo, 2021-07-19:
### Is your feature request related to a problem? Please describe.

Not a problem with OctoPrint, but in general the hotend sensor/wiring on my printer likes to break or disconnect, so I monthly get 0°C readings from the hotend sensor.
I guess this feature can also be used by users who want a stricter monitoring of the temperatures.

### Describe the solution you'd like

An user-configurable "operative temperature range", for all sensors. Actions on out-of-range temperatures should also be configurable (ignore but log - abort print - disconnect from printer). The user should also be able to choose whether the monitoring is always on or only during prints.
As an example, I'd set my hotend range from 1°C to 250°C - If the temperature is outside this range, OctoPrint will abort the current print and send an alert (that will be catched by plugins, I guess)
A "delta-mode" would also be a nice addition, where instead of absolute temperatures, the change between two monitoring polls are observed. For example, if the temperature changes more of 50°C between two scans (which, by default, is 2 seconds), the alerts would trigger.

### Describe alternatives you've considered

To my understanding, the firmware on the printer should actually send an alert if the temperature drop out-of-range. I guess it doesn't.

### Additional context

Wanhao duplicator i3 plus
OctoPrint 1.6.1 Python 3.7.3 OctoPi 0.17.0

cp2004, 2021-07-21:
This is something that is done by the firmware of the printer. Most printers have 'Max temp' and 'Min temp' settings, which would kill the printer if it is reported outside the range. If your firmware doesn't implement these (which it should - max and min are basic safety features) then you should find an alternative source of firmware for your printer, that does have safety features.

This could quite easily be done as a plugin, and there is already one for thermal runaway detection anyway. As I was typing that I found the link to the plugin (https://plugins.octoprint.org/plugins/ThermalRunaway/) and realised it does max and min temperatures as well, covering your issue well.

So, there is a plugin available for this issue, so it can be closed.

closed 2021-07-21 by cp2004

ElDavoo, 2021-07-26:
Thank you, for some reason i couldn't find that plugin, sorry for the waste of time