OctoPrint/OctoPrint #4189: Octoprint 1.6.1 goes in pause mode without any error

kami83, 2021-07-20:
### The problem

Hi,

after updating to 1.6.1 my octoprint goes on different steps in the same gcode file in to Pausing mode:

2021-07-20 13:08:23,152 - Recv: ok
2021-07-20 13:08:23,158 - Send: N222160 G1 X85.245 Y62.277 E1822.14645*100
2021-07-20 13:08:23,229 - Recv: ok
2021-07-20 13:08:23,239 - Send: N222161 G1 X85.245 Y61.968 E1822.15572*101
2021-07-20 13:08:23,250 - Recv: ok
2021-07-20 13:08:23,257 - Send: N222162 G1 X91.538 Y55.675 E1822.42287*101
2021-07-20 13:08:23,317 - Recv: ok
2021-07-20 13:08:23,323 - Send: N222163 G1 X97.754 Y55.675 E1822.60945*111
2021-07-20 13:08:23,332 - Changing monitoring state from "Printing" to "Pausing"
2021-07-20 13:08:23,433 - Recv: ok
2021-07-20 13:08:23,437 - Send: N222164 M400*22
2021-07-20 13:08:23,779 - Recv:  T:195.43 /195.00 B:60.00 /60.00 @:69 B@:61
2021-07-20 13:08:25,006 - Recv: ok
2021-07-20 13:08:25,018 - Send: N222165 M114*23
2021-07-20 13:08:25,029 - Recv: X:97.75 Y:55.68 Z:3.64 E:1822.61 Count X:7740 Y:4454 Z:1987
2021-07-20 13:08:25,245 - Recv: ok
2021-07-20 13:08:25,262 - Changing monitoring state from "Pausing" to "Paused"
2021-07-20 13:08:25,787 - Recv:  T:195.26 /195.00 B:60.00 /60.00 @:70 B@:60
2021-07-20 13:08:27,776 - Recv:  T:195.07 /195.00 B:60.01 /60.00 @:72 B@:58
2021-07-20 13:08:29,774 - Recv:  T:194.74 /195.00 B:60.00 /60.00 @:75 B@:60
2021-07-20 13:08:31,773 - Recv:  T:194.47 /195.00 B:60.00 /60.00 @:78 B@:60
2021-07-20 13:08:33,772 - Recv:  T:194.34 /195.00 B:60.00 /60.00 @:78 B@:60
2021-07-20 13:08:35,770 - Recv:  T:194.28 /195.00 B:60.00 /60.00 @:78 B@:60

I don't know why. How can i find out why Octoprint is doing this??

Thanks a lot.

BG kami

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

It happens on different steps

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Anycubic Mega 

### Browser and version of browser, operating system running browser

Chromium

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

foosel, 2021-07-20:
You forgot to share a system info bundle.

kami83, 2021-07-20:
[octoprint-systeminfo-20210720134857.zip](https://github.com/OctoPrint/OctoPrint/files/6848026/octoprint-systeminfo-20210720134857.zip)

kami83, 2021-07-20:
I am sorry.

cp2004, 2021-07-20:
It says in that log file that the print was paused by the plugin 'filament sensor revolutions', not a bug in OctoPrint and probably intended behaviour from the plugin?

kami83, 2021-07-20:
Hi, but after some times it get the same problem without filament sensor revolutions.

foosel, 2021-07-20:
Then we need you to reproduce the same behaviour in **safe mode** and share a system info bundle from that.

foosel, 2021-07-20:
Also, maybe unrelated, but your Pi is running throttled. Make sure it's properly cooled and the PSU is actually supplying enough juice.

kami83, 2021-07-30:
Hi,

i update to Octopi 0.18 now everything works fine.

Dont know why?

BG kami

cp2004, 2021-07-30:
If you didn't reinstall the filament sensor plugin that was causing issues before, then there will be no more issues.

closed 2021-07-30 by cp2004

kami83, 2021-07-30:
No, i reinstall it.