OctoPrint/OctoPrint #4194: Crash due to _self.resendDelta likely not set.

dontworryboutit, 2021-07-25:
### The problem

Just installed the MMU2S on my prusa. I made the changes to the octoprint printer profile extruder to add extruders and check the shared nozzle checkbox. Since I installed the MMU2S I've had a few issues but I attributed them up to calibration. Then suddenly, mid-print, the printer indicated that there was something wrong with the MMU2S and that I need to fix it. Then I check the octoprint web gui and it seems like the serial connection to the printer was aborted. I've never seen this before in the many hours of print I've done with this specific installation of octoprint. Looked in `octoprint.log` and found this traceback. 
 

```021-07-25 13:36:23,470 - octoprint.plugins.thespaghettidetective - DEBUG - Sending to server: 
{'octoprint_data': {'state': {'text': 'Cancelling', 'flags': {'operational': True, 'printing': True, 'cancelling': True, 'pausing': False, 'resuming': False, 'finishing': False, 'closedOrError': False, 'error': False, 'paused': False, 'ready': False, 'sdReady': True}, 'error': ''}, 'job': {'file': {'name': 'monkey_mmu2s_print_0.2mm_PETG_MK3SMMU2S_2h47m.gcode', 'path': 'monkey_mmu2s_print_0.2mm_PETG_MK3SMMU2S_2h47m.gcode', 'display': 'monkey_mmu2s_print_0.2mm_PETG_MK3SMMU2S_2h47m.gcode', 'origin': 'local', 'size': 3620503, 'date': 1627228180}, 'estimatedPrintTime': 10191.104086486888, 'averagePrintTime': None, 'lastPrintTime': None, 'filament': {'tool0': {'length': 6957.707064067879, 'volume': 16.73524669557762}}, 'user': 'cokeaddict56'}, 'currentZ': None, 'progress': {'completion': None, 'filepos': None, 'printTime': None, 'printTimeLeft': None, 'printTimeLeftOrigin': None}, 'offsets': {}, 'resends': {'count': 7, 'transmitted': 113878, 'ratio': 0}, 'temperatures': {'tool0': {'actual': 110.2, 'target': 225.0, 'offset': 0}, 'tool1': {'actual': 110.2, 'target': 225.0, 'offset': 0}, 'tool2': {'actual': 110.2, 'target': 225.0, 'offset': 0}, 'tool3': {'actual': 110.2, 'target': 225.0, 'offset': 0}, 'tool4': {'actual': 110.2, 'target': 225.0, 'offset': 0}, 'bed': {'actual': 90.1, 'target': 90.0, 'offset': 0}, 'chamber': {'actual': None, 'target': None, 'offset': 0}}, '_ts': 1627238183, 'file_metadata': {'analysis': {'printingArea': {'maxX': 240.0, 'maxY': 121.45700073242188, 'maxZ': 101.4000015258789, 'minX': 5.0, 'minY': -3.0, 'minZ': 0}}}}, 'current_print_ts': -1}
2021-07-25 13:36:25,169 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-07-25 13:36:28,178 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-07-25 13:36:31,188 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-07-25 13:36:34,198 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-07-25 13:36:37,208 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-07-25 13:36:40,217 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-07-25 13:36:43,227 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-07-25 13:36:46,237 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-07-25 13:36:49,247 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-07-25 13:36:52,257 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-07-25 13:36:55,268 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-07-25 13:36:56,487 - octoprint.util.comm - INFO - Communication timeout during an active resend, resending same line again to trigger response from printer.
2021-07-25 13:36:56,488 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 2465, in _monitor
    self._handle_timeout()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 3145, in _handle_timeout
    if self._resendSameCommand():
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 4140, in _resendSameCommand
    return self._resendNextCommand(again=True)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 4163, in _resendNextCommand
    cmd = self._lastLines[-self._resendDelta].decode("ascii")
IndexError: deque index out of range
2021-07-25 13:36:56,492 - octoprint.events.fire - DEBUG - Firing event: Error (Payload: {'error': 'See octoprint.log for details', 'reason': 'crash'})
2021-07-25 13:36:56,492 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x9e1c6cd0>>
```

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.6.1

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Prusa MK3S+ MMUS2 Firmware 3.10.0 (latest)

### Browser and version of browser, operating system running browser

Brave

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20210725152031.zip](https://github.com/OctoPrint/OctoPrint/files/6874543/octoprint-systeminfo-20210725152031.zip)
[monkey_mmu2s_print_0.2mm_PETG_MK3SMMU2S_2h47m.gcode.zip](https://github.com/OctoPrint/OctoPrint/files/6874544/monkey_mmu2s_print_0.2mm_PETG_MK3SMMU2S_2h47m.gcode.zip)
[octoprint.log.zip](https://github.com/OctoPrint/OctoPrint/files/6874553/octoprint.log.zip)

foosel, 2021-07-26:
Something happened during a resend request from your printer triggering this. But with neither `serial.log` nor a terminal dump in the bundle I fear I have nothing to go on here to analyse this further. I can take a look at the code and see how it could harden it against such a scenario, but without knowing what even led to it that is also just poking at it with a stick pretty much.

I'll mark this as unreproduced bug and requiring more information for now, and would request that whoever runs into this next provide at the very least the contents of the the terminal tab so we might get a rough idea of what happened leading to this error.

dontworryboutit, 2021-07-26:
@foosel  If I'm able to reproduce, I'll definitely try to provide the information you requested. I wasn't sure if this function as supposed to exception out or if it could be handled more gracefully. But like yourself, I'd have to dig through the code to see where/how it makes sense to handle this exception. Mainly wanted to raise awareness. Thanks!

foosel, 2021-07-26:
I've added some slightly better error handling ("insufficient history" will now trigger also for retriggered resends). That's not a fix, but it should at least no longer crash the send loop and hopefully also give some more debug output. Will be in 1.7.0.

dontworryboutit, 2021-07-26:
@foosel  I added a comment to your commit with regard to setting `self._resendDelta = None`, as I'm not as familiar with the code base, feel free to disregard if I'm missing something.

https://github.com/OctoPrint/OctoPrint/commit/52e105416d69ba14cbddf269e5c19f13d1271823#r53986850