OctoPrint/OctoPrint #4195: [Request] backup plugin hook for additional_incudes

jneilliii, 2021-07-26:
### Is your feature request related to a problem? Please describe.

It would be nice to be able to backup files outside of OctoPrint's default base folder. 

### Describe the solution you'd like

Not sure how often it comes up for other plugins, assuming they store files in the plugin's data folder, but it might be helpful to have an `additional_includes` hook in the backup plugin similar to [octoprint.plugin.backup.additional_excludes](https://docs.octoprint.org/en/master/bundledplugins/backup.html#octoprint-plugin-backup-additional-excludes) that will also restore to original path location. For my use case it's related to OctoDash's config.json and custom theme's css file that is stored in it's own path in the user's home folder. I'd like to be able to include those in backups with OctoDash Companion. Might be useful for other plugins that reference external user created system scripts as well. 

### Describe alternatives you've considered

Keeping a copy of the files during upload to the plugin's data folder, and handling during restore (don't see a backup restored event though to react to). Only issue there is if they are modified outside of OctoPrint (ie from OctoDash directly).

foosel, 2021-11-10:
What I'm worried about here is 

1. This getting abused by plugin authors either intentionally or by accident including half the file system, rendering the backup unusable or outright too big to render. We could potentially combat that with some sort of size limitation for additional includes maybe, and some heavy filtering of acceptable paths (e.g. forbidding `/dev`, `/var`, `/proc`, `/sys` and `/etc` outright).
2. Permissions are tricky here. Everything that is currently being backed up belongs to the OctoPrint process, and it should always be able to read and write it. If this gets opened up to other files, we might try to restore files we can't write (e.g. a `printer.cfg`). How to proceed in that case? Declare the backup failed? Handle it similarily to plugins that couldn't be reinstalled?
3. The mapping for this will be ugly. Right now OctoPrint's backup structure pretty much mirrors its own data folder. That makes things easy. For anything outside of that structure we'd need to add some sort of mapping information file. 

I'm wondering if this isn't rather a case of making it easier or just clearly documenting on how to use an established backup tool to back up both OctoPrint (e.g. by running *its* backup first) instead of trying to turn OctoPrint's internal backup plugin into one.

jneilliii, 2021-11-10:
I completely understand the concerns, but could utilize the proposed alternative of keeping a copy of the file on save in the plugin's data folder, but even then it would require some form of hook for `on_after_restore`, which based on the docs doesn't seem to exist. In my case for OctoDash, the files are owned by the same user as OctoPrint as they are contained within the user's home directory so `#2` doesn't apply, I can't speak on the klipper scenario.

alexbarcelo, 2022-01-03:
I am a Klipper user, and I was looking for this `additional_includes`, until I realized (when I found this issue) that it didn't exist.

I wholeheartedly agree with foosel concerns. I know that I wouldn't shoot myself in the foot, but users / plugin authors might perform unreasonable actions and render backups unusable --and that is detrimental to a lot of users. I believe that only allowing paths inside the home would be reasonable (and prevent lots of disasters). There is still room for misuse, but less likely IMHO. Regarding permissions, `printer.cfg` is by default in the pi's home (no permission problem expected there) and I believe OctoDash also falls in this scenario.

wernight, 2022-08-18:
Another way would be to be like HomeAssistant where the `/config` contains all and only the configs. Almost like the current home, just excluding for example pipenv, pip cache...

foosel, 2022-11-17:
Thought a lot about this again when I went over it while processing the 1.9.0 backlog, and I think I've found a good compromise.

I still don't feel comfortable to create a dedicated `additional_files` with OS mapping and whatnot. BUT I added four new hooks:

- octoprint.plugin.backup.before_backup
- octoprint.plugin.backup.after_backup
- octoprint.plugin.backup.before_restore
- octoprint.plugin.backup.after_restore

Those will allow plugins to perform additional tasks right before/after backup and right before/after restore. And those tasks could also be copying stuff from the file system into the plugin's data folder, and moving it around after.

Quick demo:

```python
import octoprint.plugin

import os
import datetime
import shutil

class BackupHookTestPlugin(octoprint.plugin.OctoPrintPlugin):
    def before_backup(self, *args, **kwargs):
        self._logger.info("before_backup")
        with open(os.path.join(self.get_plugin_data_folder(), "test.txt"), "w") as f:
            f.write(datetime.datetime.now().isoformat())

    def after_backup(self, error=False, *args, **kwargs):
        self._logger.info(f"after_backup, error={error}")
        os.remove(os.path.join(self.get_plugin_data_folder(), "test.txt"))

    def before_restore(self, *args, **kwargs):
        self._logger.info("before_restore")

    def after_restore(self, error=False, *args, **kwargs):
        self._logger.info(f"after_restore, error={error}")
        shutil.move(os.path.join(self.get_plugin_data_folder(), "test.txt"), os.path.join(self.get_plugin_data_folder(), "test.restored.txt"))

__plugin_pythoncompat__ = ">=3,<4" # only python 3
__plugin_implementation__ = BackupHookTestPlugin()
__plugin_hooks__ = {
    "octoprint.plugin.backup.before_backup": __plugin_implementation__.before_backup,
    "octoprint.plugin.backup.after_backup": __plugin_implementation__.after_backup,
    "octoprint.plugin.backup.before_restore": __plugin_implementation__.before_restore,
    "octoprint.plugin.backup.after_restore": __plugin_implementation__.after_restore,
}
```

Important to note is that the `{before|after}_restore` stuff will obviously not be called in case of a fully manual restore (so, by manually unpacking the backup zip, instead of going through OctoPrint's restore commands). So my suggestion for plugins wishing to react to a restore at all times would be to throw a small flag file into their data folder on backup (and remove right after backup), check if that is there on startup, if so handle their restore routine and delete it - in addition to implementing the `{before|after}_restore` hooks.

Will be part of 1.9.0

foosel, 2023-05-23:
1.9.0 has been released

closed 2023-05-23 by foosel