OctoPrint/OctoPrint #4196: Prompt before uploading file

9ae8sdf76, 2021-07-26:
### Is your feature request related to a problem? Please describe.

Selecting a file in the Files widget will cause a non-managed print (i.e., a print started at the printer) to be stopped mid-print. 

### Describe the solution you'd like

> A clear and concise description of what you want to happen.

1. Check the status of the printer (e.g., [M27](https://marlinfw.org/docs/gcode/M027.html) and/or [M31](https://marlinfw.org/docs/gcode/M031.html))
2. If receive back "SD printing byte X/Y" or "Print time: Xh Ym Zs" (where X > 0 or Y > 0 or Z > 0), then confirm before uploading warning users that any current print jobs will be lost
3. If receive back "Not SD printing" or "Print time: 0h 0m 0s", perform normal operations

(Alternatively, if Octoprint detects a bed/nozzle temp, assume a print job is started)

> Explain why it needs to be included in OctoPrint's core and can't become a plugin.

The precedent already exists by having the **Confirm before starting a print**, **Confirm before cancelling a print**, **Confirm before overwriting a file by upload** options under **Features**. Furthermore, it's not entirely clear to me that conditionally halting an upload is possible from the plugin API (admittedly, I could be wrong). 

### Describe alternatives you've considered

N/A

### Additional context

1. I am a new Octoprint user
2. I am a new 3D printer owner
3. The print was started from the printer, off of the SD card
4. I have the **Analyze gcode for time and model size estimates** set to **Always** (unsure if relevant)

surdu, 2021-12-31:
I didn't manage to reproduce this, at least not with the virtual printer.

What I did: 
1. started a print
2. selected another uploaded file
3. deleted that file
4. uploaded it again

After all this, the print started at step 1 was still printing.

@9ae8sdf76 have I misunderstood something ?

cp2004, 2021-12-31:
It might not work on the virtual printer, not sure. In this case the print that was interrupted was printing from the SD card on the printer (not managed by OctoPrint). The virtual printer does have a 'virtual SD' but not sure if it will work exactly the same way.

foosel, 2022-01-17:
@9ae8sdf76 Can you please [enable serial.log](), reproduce the issue, the upload a [system info bundle]() here? I have no clue right now what happened here with your print that led you to create this request in the first place, but it sounds more like a bug in either OctoPrint or the printer's firmware than a feature request, so I need more information to analyse this.

foosel, 2022-06-20:
No logs, no bundle. Can't debug thin air :/