OctoPrint/OctoPrint #4198: Plugin Manager Freezing / Gets Stuck

GeoGeo116, 2021-08-03:
### The problem

Hiya,

So I have recently tried installing plugins for octolapse to record time lapses. I have watched a lot of videos and followed them perfectly as they do it, however, when I click install, it gets stuck (see image below). The 'cancel' button doesn't work and if I click outside the install box, it just shakes. I have tried a fresh install of octopi and it still doesn't work. I have also tried installing it from the url instead of plugin browser, but the same thing happens.

Logs:
http://192.168.1.154/#term

Screenshot:
https://gyazo.com/735a997d56d572c6e666f23d0a0bf7a5

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

I can't install plugins in safe mode

### Version of OctoPrint

1.5.2

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Ender 3 Pro (Included Firmware)

### Browser and version of browser, operating system running browser

Chrome to access octoprint

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

jneilliii, 2021-08-03:
Please upload the requested system bundle. You can drag it directly into a comment after downloading from your octoprint instance.

jneilliii, 2021-08-03:
I'm going to take a wild guess that this is undervoltage issues. Future versions of OctoPrint will show better error messages for that. Otherwise it's a specific issue with the plugin and you should open a issue in that repository.

foosel, 2021-08-03:
Also update OctoPrint first please, the current version is 1.6.1, you are still on 1.5.2.

GeoGeo116, 2021-08-03:
> Please upload the requested system bundle. You can drag it directly into a comment after downloading from your octoprint instance.

hi, how do i do that?
Sorry, im new to this

GeoGeo116, 2021-08-03:
> Also update OctoPrint first please, the current version is 1.6.1, you are still on 1.5.2.

GeoGeo116, 2021-08-03:
> Also update OctoPrint first please, the current version is 1.6.1, you are still on 1.5.2.

I tried updating it and it says this:

Update not started!
The update could not be started. Is it already active? Please consult octoprint.log for details.

jneilliii, 2021-08-03:
octoprint.log can be found inside the logging section of OctoPrint.

GeoGeo116, 2021-08-03:
> octoprint.log can be found inside the logging section of OctoPrint.

this is the log after i try to update it:
https://pastebin.com/Lvc3Hs29

foosel, 2021-08-03:
@jneilliii called it, you are running worth undervoltage.

https://community.octoprint.org/t/pi-undervoltage-power-issues-i-had-a-strange-lightning-bolt-temperature-symbol-with-an-exclamation-mark-popup-in-my-navbar-what-does-that-mean/5256

Fix this first, this can cause all kinds of issues, which is why updates and plugin installs are blocked while it's active. Older OctoPrint versions still had a bug that didn't properly reflect this on the front-end however.

GeoGeo116, 2021-08-03:
> @jneilliii called it, you are running with undervoltage.
> 
> https://community.octoprint.org/t/pi-undervoltage-power-issues-i-had-a-strange-lightning-bolt-temperature-symbol-with-an-exclamation-mark-popup-in-my-navbar-what-does-that-mean/5256
> 
> Fix this first, this can cause all kinds of issues, which is why updates and plugin installs are blocked while it's active. Older OctoPrint versions still had a bug that didn't properly reflect this on the front-end however.

Ahh, makes sense. How could i fix this. Thanks

foosel, 2021-08-03:
Click the link I posted, that tells you how to fix it.

cp2004, 2021-08-04:
This is a known bug with OctoPrint 1.5.1 & 1.5.2, fixed in 1.5.3. So it is a duplicate of #3849 & once the undervoltage issue is resolved then the plugin manager & SW update both work normally.

closed 2021-08-04 by cp2004