OctoPrint/OctoPrint #4199: Data corruption with gcode files

jadamsowers, 2021-08-04:
### The problem

I have been experiencing some weird behavior with my Prusa Mk3s. Sometimes a print would consistently exhibit the same layer shifting at the same height, sometimes it would stop with an M112 error message, etc. I thought I was having hardware issues until I decided to download and scan through the gcode around where an error was happening. 

What I saw was weird corruption of the gcode file itself right around where the error was happening in this particular print. I have included some example files, but here is a line that changed somewhere from prusaslicer to octoprint: 

```shell
G1 X147.729 Y103.¸'ëlv´@¤ñ<ü†Ý` ´@&Â ?Ü[–·c‡¡Ä&Â ?Î†Í¦­™”â Pû‹Üm¼€ ¥C  
²ØàSn}(X103.484 Y106.907 E2.87477
```

I usually use the prusaslicer gcode export to octoprint, but I have also found this issue by saving the file locally and then uploading to octoprint and immediately redownloading it, both in regular and safe mode. I also installed OctoPi to a different SD card to see if there was an issue with the original SD storage. 

[TFSD_CHolder_Single_0.3mm_PLA_MK3S_1h7m-prusaslicer.gcode.txt](https://github.com/OctoPrint/OctoPrint/files/6928687/TFSD_CHolder_Single_0.3mm_PLA_MK3S_1h7m-prusaslicer.gcode.txt)
[TFSD_CHolder_Single_0.3mm_PLA_MK3S_1h7m-octoprint.gcode.txt](https://github.com/OctoPrint/OctoPrint/files/6928688/TFSD_CHolder_Single_0.3mm_PLA_MK3S_1h7m-octoprint.gcode.txt)
[TFSD_CHolder_Single_0.3mm_PLA_MK3S_1h7m-octoprint-safemode.gcode.txt](https://github.com/OctoPrint/OctoPrint/files/6928689/TFSD_CHolder_Single_0.3mm_PLA_MK3S_1h7m-octoprint-safemode.gcode.txt)

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Prusa Mk3s

### Browser and version of browser, operating system running browser

Brave V1.27.109 MacOS

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20210804010109.zip](https://github.com/OctoPrint/OctoPrint/files/6928717/octoprint-systeminfo-20210804010109.zip)
[image]
(An example of the gcode viewer attempting to render one of the corrupted lines (layer number 172))

jadamsowers, 2021-08-05:
Update: I remembered that my CNC machine is also connected to a different RPi running Octoprint and had also been behaving strangely. I checked some of the .gcode on that machine and it is also corrupted, although more 'silently'... no weird ASCII garbage, but invalid commands like
```
G01 X467.3495 Y73.7919 F1200
G01200
```
where obviously everything from `X` to `F` was missing. Another instance where at least some data is missing (perhaps a `0.` in the second line between `Y12` and `0519`?) sending my Y to `120519` (!):
```
G01 X517.1775 Y119.6034 F1200
G01 X517.0628 Y120519 F1200
```

This is different hardware, different SD cards, different apps generating gcode (Prusaslicer for my 3d printer, ESTLcam for my CNC) but both are RPis running the latest Octoprint. 

[2-FdL.gcode.txt](https://github.com/OctoPrint/OctoPrint/files/6940509/2-FdL.gcode.txt)

cp2004, 2021-08-05:
This issue will not be ignored, it's just annoying timing as the 1.7.0 release candidate cycle started yesterday. So it will get looked at *at some point*.

After a brief look at the files you've sent across all I can confirm is that the issue most definitely exists in those files but at the same time I have no idea why or how that might happen... Will have to wait for someone smarter than me :)

foosel, 2021-11-10:
No intent to ignore, but frankly no idea where to even start debugging this. I've seen some GCODE corruption happen in the past, but that was when I still had race conditions in the upload handler and those were quite easy to trigger. This *sounds* like filesystem corruption, but you said you already ruled this out by using a different card, so right now I'm at a loss frankly.

foosel, 2024-08-19:
Closing as this appears to have been an isolated issue as so far we've not seen any more reports about it, and it was about a by now ancient version in any case and a ton has changed since.

If this is still happening with the latest stable, please open a new issue.

closed 2024-08-19 by foosel