OctoPrint/OctoPrint #4200: ETA / Filesize not reloading after reupload (overwrite) in combination with the arcwelder plugin

schorsch3000, 2021-08-04:
### The problem

while having the arcwelder plugin installed and configured to automatically run after upload, then preforming the following steps:

 - uploading a file
 - waiting for arcwelter to do its thing
 - printing that file
 - uploading a file with the same name (eg. fine print after the test was successfully) but a different size and choosing overwrite
 - clicking print before ETA is recalculated

will result in the ETA and the print percentage be calculated from the actual file position but from the original file metadata:
eg: a print running up to 300% finish.

this does not happen if there is enough time between uploading and start printing now if i open another file and that file again.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

the arcwelder plugin needs to be involved

### Version of OctoPrint

OctoPrint 1.6.1 Python 3.7.3 

### Operating system running OctoPrint

OctoPi 0.14.0

### Printer model & used firmware incl. version

Anet board running marlin, but thats quite sure irrelevant

### Browser and version of browser, operating system running browser

Vivaldi	4.0.2312.41 (Stable channel) stable (64-bit)  / Debian 10

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

browser.user_agent: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.166 Safari/537.36
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1200
env.hardware.ram: 904646656
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 3 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.14.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.1.1
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.6.1
printer.firmware: Marlin 2.0.7.2 (Mar 11 2021 21:18:38)
systeminfo.generator: systemapi
[octoprint-systeminfo-20210804133815.zip](https://github.com/OctoPrint/OctoPrint/files/6930895/octoprint-systeminfo-20210804133815.zip)

apart from "adding buttons for file with hash" there is nothing in the dev-console

this is what it looks like:
[image]

and here the 2 files to replicate the issue:

first:
```
G28
G1 F1000
G1 Z10
G1 X100
G1 X0
```

second: 
```
G28
G1 F1000
G1 Z10
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
G1 X100
G1 X0
```

schorsch3000, 2021-08-04:
@FormerLurker, maybe you'll have a look too?

also see: https://twitter.com/schorsch3000/status/1422870921392672771

cp2004, 2021-08-04:
What arc welder settings are you using? Do you have overwrite original file enabled, or is it saving to a different filename?

schorsch3000, 2021-08-04:
It's a curse, i've tried my best to give every little detail but oviously failed :D
i'm saving to a different file, that different file was openend while uploading the second file and AW is overwriting the new file

FormerLurker, 2021-11-10:
I will take a look on my end, but this may be solved in devel.  I'll report back.

foosel, 2023-01-24:
@FormerLurker Did you ever get a chance to look at this?

foosel, 2024-04-11:
Closing as this is out of my hands.

closed 2024-04-11 by foosel