OctoPrint/OctoPrint #4201: [RC Feedback] Feedback on the 1.7.0rc1 Release Candidate

foosel, 2021-08-04:
Please provide *general* feedback on your experience with the 1.7.0rc1 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### If you run into any obvious bugs not yet listed below the following line, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports, not just a statement that something doesn't work!

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

b-morgan, 2021-08-04:
Upgraded successfully.

jneilliii, 2021-08-04:
3 successful upgrades, printed on one without issue.

Andy-ABTec, 2021-08-05:
Upgrade ok + a couple of small prints without issue - _so far so good..._

The-EG, 2021-08-05:
Upgraded my 2 instances in one venv setup and completed a 5 hour print with no issues. Really enjoying the animation/window changes...it's the little things sometimes!

MangaValk, 2021-08-05:
```Updating, please wait.
++++++++++++++++++++++++++++++++++
Now updating OctoPrint to 1.7.0rc1
++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/OctoPrint/OctoPrint/archive/1.7.0rc1.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/OctoPrint/OctoPrint/archive/1.7.0rc1.zip
Downloading https://github.com/OctoPrint/OctoPrint/archive/1.7.0rc1.zip (6.6MB)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc1) (2021.2.23)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.2.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc1) (2021.5.31)
Requirement already satisfied: OctoPrint-PiSupport>=2021.8.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc1) (2021.8.2)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc1) (1.1.1)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc1) (3.1.1)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc1) (2019.8.19)
Collecting flask<2,>=1.1.4
Downloading https://files.pythonhosted.org/packages/e8/6d/994208daa354f68fd89a34a8bafbeaab26fda84e7af1e35bdaed02b667e6/Flask-1.1.4-py2.py3-none-any.whl (94kB)
Collecting Jinja2<3,>=2.11.3
Downloading https://files.pythonhosted.org/packages/7e/c2/1eece8c95ddbc9b1aeb64f5783a9e07a286de42191b7204d67b7496ddf35/Jinja2-2.11.3-py2.py3-none-any.whl (125kB)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc1) (0.5.0)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc1) (1.0.0)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc1) (2.0)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc1) (1.0.1)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc1) (1.1.0)
Requirement already satisfied: cachelib<0.2,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc1) (0.1)
Collecting PyYAML<6,>=5.4.1
Downloading https://files.pythonhosted.org/packages/a0/a4/d63f2d7597e1a4b55aa3b4d6c5b029991d3b824b5bd331af8d4ab1ed687d/PyYAML-5.4.1.tar.gz (175kB)
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 /home/pi/oprint/local/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-wDu2P6/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools wheel Cython
cwd: None
Complete output (24 lines):
Traceback (most recent call last):
File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
"__main__", fname, loader, pkg_name)
File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
exec code in run_globals
File "/home/pi/oprint/lib/python2.7/site-packages/pip/__main__.py", line 16, in <module>
from pip._internal.main import main as _main  # isort:skip # noqa
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/main.py", line 13, in <module>
from pip._internal.cli.autocompletion import autocomplete
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/cli/autocompletion.py", line 11, in <module>
from pip._internal.cli.main_parser import create_main_parser
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/cli/main_parser.py", line 7, in <module>
from pip._internal.cli import cmdoptions
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/cli/cmdoptions.py", line 25, in <module>
from pip._internal.locations import USER_CACHE_DIR, get_src_prefix
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/locations.py", line 18, in <module>
from pip._internal.utils import appdirs
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/utils/appdirs.py", line 16, in <module>
from pip._internal.utils.compat import WINDOWS, expanduser
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/utils/compat.py", line 17, in <module>
from pip._vendor.urllib3.util import IS_PYOPENSSL
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_vendor/urllib3/__init__.py", line 21, in <module>
from logging import NullHandler
ImportError: cannot import name NullHandler
----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/oprint/bin/python2 /home/pi/oprint/local/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-wDu2P6/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools wheel Cython Check the logs for full command output.
Installing build dependencies: started
Installing build dependencies: finished with status 'error'
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.```

First update ever that gives an error to me. I'm in no hurry to update, but sharing this for your information.

cp2004, 2021-08-05:
@MangaValk take a look here: https://community.octoprint.org/t/my-octoprint-update-fails-with-importerror-cannot-import-name-nullhandler/34493

It was an issue with a PrintTimeGenius version that pulls in an incompatible 3rd party dependency, that overwrites the standard library.

It's been a while since we've seen that one but I guess since we are on the upgrade cycle again it will probably be around again.

JohnOCFII, 2021-08-06:
1.7.0rc1 seems to be working fine.

- Uploaded new GCODE to OctoPrint
- Printed GCODE on Prusa MK3S/wMMU2 in single extruder mode
- GCODE viewer worked well
- Terminal worked well
- Timelapse worked well
- The Prusa MMU2 Select Filament plugin worked well
- OctoSlack plugin worked well
- Print Job History plugin worked well
- Telegram plugin worked well

kazibole, 2021-08-08:
All is working fine - Prusa MK3S and printer's SD card prints are working fine.

loskexos, 2021-08-09:
All prints since upgrade 2 days ago fine. 
(sent from prusaslicer to octoprint. plugins active: multicam, tempsgraph, slicer thumbnails, prusa levelling guide)

foosel, 2021-08-11:
1.7.0rc2 has been released, new ticket is #4206

closed 2021-08-11 by foosel