OctoPrint/OctoPrint #4202: Error while estimating print time

jrydval, 2021-08-05:
### The problem

The UI doesn't show Print Time Left at the beginning of a print. The log is full of error messages like this one:

```
2021-08-05 19:18:42,052 - octoprint.printer.standard - ERROR - Error while estimating print time via <octoprint.printer.estimation.PrintTimeEstimator object at 0xab532ed0>
Traceback (most recent call last):
  File "/home/pi/Apps/OctoPrint/venv/lib/python3.7/site-packages/octoprint/printer/standard.py", line 1194, in _updateProgressDataCallback
    printTimeLeft = int(ptl)
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
```

The number of these errors is:
# cat .octoprint/logs/octoprint.log | grep "Error while estimating" | wc -l
557

After that is starts working fine and Print Time Left is displayed as usual.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Just printing a longer project

### Version of OctoPrint

1.7.0rc1

### Operating system running OctoPrint

Raspbian GNU/Linux 10

### Printer model & used firmware incl. version

MK3S+

### Browser and version of browser, operating system running browser

Safari, MacOS, BigSur

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20210805192956.zip](https://github.com/OctoPrint/OctoPrint/files/6940569/octoprint-systeminfo-20210805192956.zip)

jrydval, 2021-08-05:
Actually I am questioning the fact that Print Time Left is empty at the beginning. Just the ERRORs in the log file... Are they really needed?

cp2004, 2021-08-05:
Possibly a regression, not sure yet as I haven't tried it. But what I can see:

* There's file analysis started:
  ```
  2021-08-05 19:12:37,070 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Filament_Guide_Clip_Angle_+_Filament_Store_0.2mm_PETG_MK3S_1h5m.gcode
  2021-08-05 19:12:37,072 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/Apps/OctoPrint/venv/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /home/pi/.octoprint/uploads/Filament_Guide_Clip_Angle_+_Filament_Store_0.2mm_PETG_MK3S_1h5m.gcode
  ```
* After that, there's a print started, but analysis hasn't finished. This would be where the fact that these things are none comes from - analysis data is empty. Out of the box, analysis is usually disabled during a print so it may have stopped at that point as well.
* Then the errors come - and no, these are actual errors... OctoPrint shouldn't be trying to grab the data that it should know doesn't exist.

I know the analysis was worked on in 1.7.0 to produce empty results, it is just deciding if this is a regression or not and needs to be fixed for 1.7.0rc2.

jneilliii, 2021-08-05:
Was the file sent direct from PrusaSlicer with the option to auto start the print? I've had some weird things in that situation with OctoDash and auto powering on printer (TPLink), and logs seem to indicate something that may be similar where analysis starts and then the print kicks in before it's completed.

foosel, 2021-08-09:
Definitely a regression, the estimator may return `None` there but I didn't take that into account when adding the new type conversion. Fixing

foosel, 2021-08-09:
Fix ready for 1.7.0rc2

foosel, 2021-08-11:
1.7.0rc2 has been released

commit 2f266f99 (2021-08-09): 🐛 printTimeLeft may be None

So we need to check for that first before we try to
convert to an int.

Regression.

Closes #4202
closed 2021-08-11 by foosel