OctoPrint/OctoPrint #4203: "Could not set low latency mode on serial port, continuing without" ERROR on RPi

jrydval, 2021-08-05:
### The problem

The octoperint.log contains ERRORs like:

`2021-08-05 16:41:10,513 - octoprint.util.comm - ERROR - Could not set low latency mode on serial port, continuing without`

Setting lowLatency: false in serial section of the octoprint.yaml doesn't help.

This could seems as a cosmetic problem, but my nagios monitoring is reporting any ERROR line in octoprint.log...

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### Version of OctoPrint

1.7.0rc1

### Operating system running OctoPrint

Raspbian GNU/Linux 10

### Printer model & used firmware incl. version

MK3S+

### Browser and version of browser, operating system running browser

Safari, MacOS, BigSur

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20210805192956.zip](https://github.com/OctoPrint/OctoPrint/files/6940811/octoprint-systeminfo-20210805192956.zip)

cp2004, 2021-08-05:
Should probably be a warning log level - not setting low latency just means it behaves as it was pre-1.7.0. The setting not working is interesting though.

cp2004, 2021-08-05:
The setting still tries to set whatever it can on the serial port, so it is intentional that it won't make a difference in this case.

See few lines below this point:

https://github.com/OctoPrint/OctoPrint/blob/1fdd0a7ee17da7fe737512d830c8fbd52a7855e4/src/octoprint/util/comm.py#L3755

jrydval, 2021-08-05:
Thank you for your really quick reaction! Actually I commented out the whole section starting with "if hasattr(serial_obj, "set_low_latency_mode"): " line to get rid of the ERROR messages. Yep, INFO w/o stack trace would be enough.

jrydval, 2021-08-05:
What about to change the order of the conditions - if lowLatency is false then do not try to set it.

OR

What about to make lowLAtency set of values "false", "true", "doNotSet"?

foosel, 2021-08-09:
> What about to change the order of the conditions - if lowLatency is false then do not try to set it.

That's what we are going to do.

foosel, 2021-08-09:
Fix ready for 1.7.0rc2

foosel, 2021-08-11:
1.7.0rc2 has been released

commit a7d4c5f2 (2021-08-09): 🐛 Change latency mode setting logic

We should only ever attempt to set it if we are supposed
to set it to True. Otherwise there's no need to do
anything really and risk running into issues.

Closes #4203
closed 2021-08-11 by foosel