OctoPrint/OctoPrint #4208: Prusa MK3S+ crash causes disconnect

brett19, 2021-08-11:
### The problem

While printing recently, it appears the MK3S+ spuriously detected a crash (there was nothing obviously wrong with the print when I eventually discovered this).  Following a crash it appears that the printer goes through a quick re-check before continuing to print but Octoprint handles the recovery poorly and causes the printer to simply disconnect and an M112 to be sent.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

It's not an easily reproducible problem.  The logs also point to this not being an issue due to a plugin.

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Original Prusa MK3S+

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Log of failure:
```
Send: G1 X228.064 Y30.901 E0.27978
Recv: ok
Send: G1 X226.662 Y33.329 E0.09491
Recv: echo:enqueing "G1 E-1.000 F2700"
Recv: echo:enqueing "G1 Z15.600 F800.000"
Recv: echo:enqueing "CRASH_DETECTEDY"
Recv: ok
Send: G1 X239.762 Y40.882 E0.51186
Recv: T:209.7 /210.0 B:59.9 /60.0 T0:209.7 /210.0 @:68 B@:48 P:0.0 A:43.8
Recv: echo:busy: processing
Recv: T:209.8 /210.0 B:59.8 /60.0 T0:209.8 /210.0 @:67 B@:49 P:0.0 A:44.1
Recv: echo:busy: processing
Recv: T:211.0 /210.0 B:59.9 /60.0 T0:211.0 /210.0 @:49 B@:38 P:0.0 A:43.9
Recv: echo:busy: processing
Recv: T:211.4 /210.0 B:59.9 /60.0 T0:211.4 /210.0 @:45 B@:32 P:0.0 A:44.0
Recv: LCD status changed
Recv: tmc2130_home_enter(axes_mask=0x01)
Recv: echo:busy: processing
Recv: T:212.1 /210.0 B:60.0 /60.0 T0:212.1 /210.0 @:35 B@:9 P:0.0 A:44.1
Recv: echo:busy: processing
Recv: T:212.1 /210.0 B:60.0 /60.0 T0:212.1 /210.0 @:37 B@:25 P:0.0 A:44.0
Recv: echo:busy: processing
Recv:   0 step=46 mscnt= 736
Recv: tmc2130_goto_step 0 29 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
Recv: T:211.7 /210.0 B:60.0 /60.0 T0:211.7 /210.0 @:46 B@:0 P:0.0 A:44.0
Recv: tmc2130_home_enter(axes_mask=0x02)
Recv:   0 step=18 mscnt= 296
Recv: tmc2130_goto_step 1 33 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
Recv: echo:busy: processing
Recv: echo:enqueing "CRASH_RECOVER"
Recv: ok
Send: G1 X239.752 Y55.563 E0.49693
Recv: T:211.2 /210.0 B:60.2 /60.0 T0:211.2 /210.0 @:52 B@:0 P:0.0 A:43.9
Recv: echo:busy: processing
Recv: T:210.5 /210.0 B:60.3 /60.0 T0:210.5 /210.0 @:62 B@:0 P:0.0 A:44.0
Recv: echo:busy: processing
Recv: T:210.0 /210.0 B:60.1 /60.0 T0:210.0 /210.0 @:67 B@:17 P:0.0 A:44.0
Recv: echo:busy: processing
Recv: T:209.6 /210.0 B:60.2 /60.0 T0:209.6 /210.0 @:69 B@:12 P:0.0 A:43.8
Recv: echo:busy: processing
Recv: T:210.4 /210.0 B:60.1 /60.0 T0:210.4 /210.0 @:55 B@:35 P:0.0 A:43.9
Recv: echo:busy: processing
Recv: T:211.3 /210.0 B:60.1 /60.0 T0:211.3 /210.0 @:41 B@:27 P:0.0 A:44.0
Recv: echo:busy: processing
Recv: T:211.2 /210.0 B:59.9 /60.0 T0:211.2 /210.0 @:43 B@:68 P:0.0 A:43.8
Recv: echo:busy: processing
Recv: T:211.4 /210.0 B:59.9 /60.0 T0:211.4 /210.0 @:40 B@:53 P:0.0 A:43.9
Recv: echo:busy: processing
Recv: T:211.1 /210.0 B:59.9 /60.0 T0:211.1 /210.0 @:45 B@:52 P:0.0 A:43.9
Recv: echo:busy: processing
Recv: T:210.6 /210.0 B:59.9 /60.0 T0:210.6 /210.0 @:52 B@:50 P:0.0 A:43.9
Recv: Resend: 0
Printer keeps requesting line 0 again and again, communication stuck
Changing monitoring state from "Printing" to "Error"
Send: M112
Send: N1 M112*32
Send: N2 M104 T0 S0*35
Send: N3 M140 S0*102
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
Closing down send loop
```

System Info:
[octoprint-systeminfo-20210811161855.zip](https://github.com/OctoPrint/OctoPrint/files/6971869/octoprint-systeminfo-20210811161855.zip)

cp2004, 2021-08-13:
First, try uninstalling the flashforge plugin. You don't have a flashforge printer here, you have a Prusa MK3S. You may have to reset the settings - a list of the serial connection settings the plugin modifies can be found here: https://github.com/Mrnt/OctoPrint-FlashForge#settings

It appears that the flashforge plugin disables sending with checksums, since those printers don't use them. I think this is the root cause of your issue, because without line numbers and checksums then there is no way to recover from a resend request like this. The printer always thinks it is at line 0 apparently, so OctoPrint can't match which line it is supposed to resend to execute after the crash.

brett19, 2021-08-13:
Hey @cp2004 ,

I apologize for the confusion.  I previously had a flashforge printer and when I switched to the MK3S+ I wiped out the .octoprint folder, expecting to get a 'default install' (which at first glance appeared to have worked).  Looks like it still left the previous plugins installed, I'll try to remove that plugin and hopefully it doesn't happen again.  Do you know if removing the plugin would be sufficient to undo any changes it made, or do I need to reset the configuration again to get back to a standard install.

Cheers, Brett

jneilliii, 2021-08-13:
Uninstalling the plugin with the option to clean data will typically remove everything associated with a plugin.

brett19, 2021-08-13:
Thanks @jneilliii, will try that!

closed 2021-08-13 by brett19