OctoPrint/OctoPrint #4209: Support for none standard M27 printing status update

TonyKim87, 2021-08-12:
### The problem

I use Flashforge Finder 3D Printer
And use FlashForge Plugin to use that

I do SD card printing.
So the problem is Flashforge Finder response to M27 is like below

SD printing byte : XXX / 1000

So problem is my guess, Flashforge Finder is not a standard Gcode printer but OEM made one
As I understand, Standard M27 response like below

SD printing byte 123/12345

So integer after slash should be total file size, but Flashforge Finder send it as fixed 1000. And before slash integer is actually progress % * 10

My suggestion is to change like below in OctoPrint/src/octoprint/util/comm.py 

`if self._currentFile.size == 0:
                                            self._currentFile.size = total`

to just

`                                           self._currentFile.size = total`

I think it is Flashforge's fault, but isn't it better way to cover various of printer?

Hope to find your opinion

Thanks

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Flashforge Plugin itself makes to communicate to my printer

### Version of OctoPrint

1.6.1:Stable

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Flashforge Finder

### Browser and version of browser, operating system running browser

N/A

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

foosel, 2021-08-12:
That is rather something that should be solved via a plugin. I do not want to add exceptions for firmware that does interpret the established de-facto protocol like it wants, just adds to the already tall maintenance burden and simply doesn't scale given the amount of printer vendors out there that tend to try to get away with breaking protocol.

The Flashforge plugin could probably be slightly adapted to rewrite this message on the fly before OctoPrint see it via the [`octoprint.comm.protocol.gcode.received` hook](https://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-gcode-received), and since you need that anyhow as it works as the compatibility layer here, I'd suggest to file this as a request there.

closed 2021-08-12 by foosel

TonyKim87, 2021-08-12:
> That is rather something that should be solved via a plugin. I do not want to add exceptions for firmware that does interpret the established de-facto protocol like it wants, just adds to the [...]

Makes Sense. Thank you