OctoPrint/OctoPrint #4211: [Request] Add Estimated Print time to PrintStarted event

wolfwood, 2021-08-12:
### Is your feature request related to a problem? Please describe.

This is something that is sutied to a plugin, but could use a slight enhancement to the plugin interface itself.

the big picture is a shared printer in a makerspace, where users want to know if someone is printing before they make the tirip in.  I would like to be able to see if a job is active, and how long it is estimated to take, from outside the LAN of the makerspace.

This is a shared machine so existing plugins that allow remote control, or even just take a single person's API key tied to a specific cloud service account, aren't appropriate. multiple users should be able to access this data / it should be exported in a way that is safe to make publicly available.

### Describe the solution you'd like

I'm happy to hear other aproaches, but I learned about OctoPrint-IFTTT, so I thought I'd use that.  I see that it can consume Octoprint events, and on the IFTTT side, I can stick those events somewhere, like a google calendar.

I'd be happy to use some different mechanism than IFTTT to take events and turn them into a static webpage, for example, hosted elsewhere, but it seems like the starting place is the Octoprint events. 

 However, the PrintStarted event does not include a property for the print's estimated duration.  knowing a print is active (and when it concludes or fails) is a nice start, but I'd also like to know how long it is likely to take.

### Describe alternatives you've considered

a DIY plugin that implements a custom estimate by parsing gcode, and then tacks that on to the event when sharing it externally.  but since there is already `octoprint.printer.estimation.factory` to generate custom estimates and feed them back to the rest  of the system, it would be to get the official estimate as part of the PrintStarted event.

### Additional context

OctoPrint is really cool!

jneilliii, 2021-08-12:
Seems like this would be possible to achieve by using the PrintStarted event's payload to then request the file's estimation data that should in theory be already available from the [files API](https://docs.octoprint.org/en/master/api/files.html#retrieve-a-specific-file-s-or-folder-s-information).

jneilliii, 2021-08-12:
or even better, just poll the [job status API](https://docs.octoprint.org/en/master/api/job.html#retrieve-information-about-the-current-job).

cp2004, 2021-08-13:
Only thing I would point out against using the `PrintStarted` event to estimate print time is that you are not guaranteed to have a print time by that point. If the file analysis hasn't finished then it may not have a print time to give you, which might be slightly unpredictable.

wolfwood, 2021-08-13:
i think the REST API isn't appropriate for a plugin, but I'm guessing there is an equivalent of that file API in python. I'd prefer not to be polling if it can be avoided, but you are right that job status is exactly what I would want.  if this were my personal device, logging in and setting up a script on a cronjob would be no problem, but in a shared environment, a plugin is more visible/less surprising to others who may need to maintain the setup.

regarding not knowing the print time when the print starts, that makes sense, i guess maybe the estimate should have it's own event?

cp2004, 2021-08-13:
There is the `MetaDataAnalysisFinished` event ([docs](https://docs.octoprint.org/en/master/events/index.html#file-handling)) that has the `result` field. It says it's marked for internal use, but I don't think it has changed much, so if you wanted to use it... well you couldn't be stopped. I am *fairly* confident that the `result` field would be the same content as the [Gcode Analysis information](https://docs.octoprint.org/en/master/api/datamodel.html#gcode-analysis-information) data model.

jneilliii, 2021-08-13:
It does have it's own event, `MetadataAnalysisFinished`, and there are options within OP to always run those no matter if the printer is printing or not, which may be a possible workaround. 

> I'm guessing there is an equivalent of that file API in python

Yes, that's true, which could be utilized by a plugin. `self._printer.get_current_job()` I believe. You could fire that on one of them.

https://docs.octoprint.org/en/master/modules/printer.html#octoprint.printer.PrinterInterface.get_current_job

wolfwood, 2021-08-16:
well, I am able to get `estimatedPrintTime` from `MetadataAnalysisFinished.result` but it does not match the estimate that Prusa Slicer is giving (which I read somewhere OctoPrint parses out of the gCode?) also I am hitting the issue that jneilliii mentioned where the event only comes after the print finishes as I haven't found the setting to run the task in the background while printing.

but aside from all that, octoprint displays a time estimate (which matches the slicer one and the one on the printer itself) and it has it without the delayed metadata analysis task completing, so I don't think I really want the outcome of the analysis, which is wrong anyhow, I want the value octoprint is actually using. and I think it is likely available when the PrintStarted event is issued, since otcoprint is suspending background tasks while printing.

jneilliii, 2021-08-16:
> but aside from all that, octoprint displays a time estimate (which matches the slicer one and the one on the printer itself) and it has it without the delayed metadata analysis task completing.

Not sure how that works unless you have a plugin installed. There are a few that improve those, specifically PrintTimeGenius, and that may be where I saw that option.

wolfwood, 2021-08-16:
I guess I was mistaken, the estimate was not immediately available, but appeared during the first layer. seems to be from a different source than the slicer metadata (or the delayed `MetadataAnalysisFinished`) as it is starts as a wild overestimate that converges down.

the only 3rd party plugins I see are Dashboard, Bed Visualizer, DisplayLayerProgess and OctoPrint-IFTTT.

jneilliii, 2021-08-17:
I just checked, it was the PrintTimeGenius plugin that added those options. 

[image]

But just took a quick look and it's in the default settings as well under features...

[image]

cp2004, 2021-08-17:
Now you are starting to see the complicated-ness of OctoPrint's print time estimating...

If metadata analysis is not finished by the time you start a print job, the time estimate is roughly provided based on how many bytes have been printed vs. how many are left in the file, and then it learns the speed slowly over tiem. Collecting this with a single 'event' is not really possible since this estimate changes over time as more of the file has been printed.

Gcode analysis in OctoPrint is not optimized for accuracy, so if you want accurate print times then you must use PrintTimeGenius to analyse your file. To get the slicer-estimated time you would have to use a plugin that can read out the gcode comments. Due to a large number of different formats & updates changing the comments in the gcode file, it is not feasible to implement gcode-comment parsing in OctoPrint itself.

Where do we stand with this request? Metadata analysis is the only event that is related to print time, the print time available when you 'ask' OctoPrint can't really be associated with an event that I know of.

wolfwood, 2021-08-20:
slicer sticks the print time in the file name, so i guess I'll be satisfied by that for now.

If i want something that i can use for calendar events i guess I'll go with PrintTimeGenius+allow analysis while heating/printing+`MetadataAnalyaisFinished` `result.estimatedPrintTime`

thanks everyone for your input. closing.

closed 2021-08-20 by wolfwood